TheSpaghettiDetective/obico-server #595: [BUG] django.request ERROR    Internal Server Error: /printers/

wardy277, 2022-03-14:
**Describe the bug**
Internal server error when viewing printers. Admin site works, can view add and edit printers

**To Reproduce**

1. Steps to reproduce the behavior:
2. git pull or checkout latest branch
3. run docker-compose up
4. navigate to /printers - get 500 error

Going back a few commits (verified  ones). the most recent I can find that works is: df87fbeb0ab427976c7a8949b73e68d5b817e478

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
 - OS: Ubuntu
 - Version 20.04
 - Hardware Ryzen 3600x

**Additional context**
```web_1     | django.request ERROR    Internal Server Error: /printers/
web_1     | Traceback (most recent call last):
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/core/handlers/exception.py", line 34, in inner
web_1     |     response = get_response(request)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/core/handlers/base.py", line 115, in _get_response
web_1     |     response = self.process_exception_by_middleware(e, request)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/core/handlers/base.py", line 113, in _get_response
web_1     |     response = wrapped_callback(request, *callback_args, **callback_kwargs)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/contrib/auth/decorators.py", line 21, in _wrapped_view
web_1     |     return view_func(request, *args, **kwargs)
web_1     |   File "/app/app/views/web_views.py", line 48, in printers
web_1     |     return render(request, template_name)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/shortcuts.py", line 36, in render
web_1     |     content = loader.render_to_string(template_name, context, request, using=using)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/loader.py", line 62, in render_to_string
web_1     |     return template.render(context, request)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/backends/django.py", line 61, in render
web_1     |     return self.template.render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 171, in render
web_1     |     return self._render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 163, in _render
web_1     |     return self.nodelist.render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 937, in render
web_1     |     bit = node.render_annotated(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 904, in render_annotated
web_1     |     return self.render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/loader_tags.py", line 150, in render
web_1     |     return compiled_parent._render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 163, in _render
web_1     |     return self.nodelist.render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 937, in render
web_1     |     bit = node.render_annotated(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 904, in render_annotated
web_1     |     return self.render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/loader_tags.py", line 150, in render
web_1     |     return compiled_parent._render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 163, in _render
web_1     |     return self.nodelist.render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 937, in render
web_1     |     bit = node.render_annotated(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/base.py", line 904, in render_annotated
web_1     |     return self.render(context)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/template/library.py", line 192, in render
web_1     |     output = self.func(*resolved_args, **resolved_kwargs)
web_1     |   File "/app/app/templatetags/vue.py", line 29, in bundle
web_1     |     url = static(f'{prefix}/{ext}/{bundle_name}.{ext}')
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/templatetags/static.py", line 167, in static
web_1     |     return StaticNode.handle_simple(path)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/templatetags/static.py", line 118, in handle_simple
web_1     |     return staticfiles_storage.url(path)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/contrib/staticfiles/storage.py", line 153, in url
web_1     |     return self._url(self.stored_name, name, force)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/contrib/staticfiles/storage.py", line 132, in _url
web_1     |     hashed_name = hashed_name_func(*args)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/contrib/staticfiles/storage.py", line 420, in stored_name
web_1     |     raise ValueError("Missing staticfiles manifest entry for '%s'" % clean_name)
web_1     | ValueError: Missing staticfiles manifest entry for 'frontend/js/styles.js'
web_1     | django.channels.server ERROR    HTTP GET /printers/ 500 [0.02, 192.168.86.130:40106]

querex007, 2022-03-14:
Same issue over here after upgrading to the latest master commit.
Was not able to revert back to a "working" environment when manual checkout commit df87fbeb0ab427976c7a8949b73e68d5b817e478

Reverted the whole set up with a recent backup, everything is working again (older version +/- dec 2021)

kennethjiang, 2022-03-15:
@wardy277 @querex007 Did you use the `master` branch? We now switched to using `release` branch for stable releases.

querex007, 2022-03-15:
Confirmed, release branch is working. Thanks.

closed 2022-03-15 by kennethjiang

wardy277, 2022-03-15:
Thanks for the information. I must have missed this. All working fine for me too