TheSpaghettiDetective/obico-server #613: [BUG] TSD tries to load webcam image from the wrong domain

Nordikk, 2022-04-08:
I'm using TSD behind an Apache2 reverse proxy, the domain is set as tsd.xxxxx.de:443 in Django Admin
There's no webcam image showing up, so I took a look in the Chrome console, there I can see that TSD tries to connect to
`https://xxxxx.de/media/tsd-pics/snapshots/2/latest_unrotated.jpg`, 
but should be connecting to 
`https://tsd.xxxxx.de/media/tsd-pics/snapshots/2/latest_unrotated.jpg`

I'll attach screenshots.

[image]
[image]
[image]

closed 2022-04-08 by Nordikk

mcstoufer, 2022-06-24:
This issue is still here. Is it expected that this image is not supposed to be loaded and just show the alt tag?

`<img data-v-76458791="" src="http://example.com/media/tsd-pics/snapshots/1/latest_unrotated.jpg" alt="Prusa Mk3s current image" class="tagged-jpg">`

Nordikk, 2022-06-24:
Did you set your domain name? 

You’re supposed to set it first, look at the installation instructions