TheSpaghettiDetective/obico-server #619: [BUG] `web/static_build` folder is built into the image

kennethjiang, 2022-04-10:
We have [.dockerignore](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/master/web/.dockerignore) so that the big folders that are not needed in building the image are not included in the image, when the self-hosted server is rebuilt with `docker-compose up --build -d`.

But it seems `web/static_build` is still included in the resulted docker image and made it really big. Need to investigate if the role of `.dockerignore` is misunderstood. If so, we need to figure out another way to achieve the same goal.

kennethjiang, 2022-04-16:
Confirmed `.dockerignore` actually solved this problem.

closed 2022-04-16 by kennethjiang

marc0777, 2022-05-04:
I think this issue still applies on [release](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/tree/release) since the `web` folder has been moved to `backend`, but the `.dockerignore` [there](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/blob/release/.dockerignore) still references `web/static_files`.

kennethjiang, 2022-05-05:
Good point. This bug will be fixed on `release` branch with our next release, which will be soon.