TheSpaghettiDetective/obico-server #626: [BUG] Can't seek the timelapse (self-hosted server only. Chromium-based browsers only)

kennethjiang, 2022-04-16:
https://discord.com/channels/614543405724205137/964592638005235722

We've found out that this happens on chromium-based browsers on all platforms (including Android)
This also probably isn't a recent bug, because it has been happening since I installed TSD (more than half a year now)
And also, we didn't test it on the official TSD, only on our private instances

lexxxel, 2024-01-26:
Disclamer: I'm not a JS developer.
I debugged a bit and found that something goes wrong in `video.cjs.js:21722` when getting the `currentTime`. The value will be 0 if `setCurrentTime` has been called before.
```js
Player.prototype.currentTime = function currentTime(seconds) {
    if (typeof seconds !== 'undefined') {
      if (seconds < 0) {
        seconds = 0;
      }
      this.techCall_('setCurrentTime', seconds);
      return;
    }

    // cache last currentTime and return. default to 0 seconds
    //
    // Caching the currentTime is meant to prevent a massive amount of reads on the tech's
    // currentTime when scrubbing, but may not provide much performance benefit afterall.
    // Should be tested. Also something has to read the actual current time or the cache will
    // never get updated.
    this.cache_.currentTime = this.techGet_('currentTime') || 0;
    return this.cache_.currentTime;
  };
```

When you click to scrub, the `currentTime` value will be set by the mouseMove event, but `setCurrentTime` will raise an exception and not set the `currentTime` property correctly. Later, on the second call (see image below) it will return 0:
[image]
[image]
[image]

[image]
[image]
See `Watch` in the top right: currentTime = 0:
[image]

[image]

kennethjiang, 2024-01-26:
Interesting. Do you know the reason why it's reset to 0? If so, can you fix it in a PR. Please remember to open the PR toward the `master` branch, not the `release` branch.

lexxxel, 2024-01-26:
I think it's never set on the first place. It looks like an upstream problem to me, but like I said, I'm not a JS developer. I can't test updating the dependency in obico, especially because it seems that the next version (6.x) is a breaking change. They even renamed the package.

dvystrcil, 2024-02-20:
Hi folks, I am new to Obico. I built it out on my K8s and for the most part, it works. I have yet to have a spaghetti mess on my hands to test it out, but I am finding a couple of functions are not working.

The server does not have a tsd-timelapses folder. It does have tsd-pics, and I have done prints that take longer than the minimum threshold.

The screenshot is from my server container and I am reading about the pathing here
https://obico.io/docs/server-guides/ops/#access-timelapses-stored-on-your-server

I know that location is writable since I see it in the logs. I can also touch a new file in that folder.

What should I be looking for to diagnose this further?
[image]

[image]