TheSpaghettiDetective/obico-server #630: TheSpaghettiDetective-web-1 keeps restarting

Erie1, 2022-04-20:
I have been trying to run a private TSD server on Truenas scale using [truecharts docker compose](https://truecharts.org/manual/Quick-Start%20Guides/13-docker-compose/) however it didn't work, I noticed the ***-web-1 container kept restarting. I then tried to run it on my windows machine using docker desktop with WSL 2 and I'm getting the same problem, I already tried running it as administrator, as was suggested in another issue, but that didn't make a difference.

As a result I can't access TSD from neither octoprint nor from my phone or pc.

Below I have attached a trace from the docker desktop instance.
```
Operations to perform:

  Apply all migrations: account, admin, app, auth, channels_presence, contenttypes, sessions, sites, socialaccount

Running migrations:

Traceback (most recent call last):

  File "/usr/local/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute

    return self.cursor.execute(sql, params)

  File "/usr/local/lib/python3.7/site-packages/django/db/backends/sqlite3/base.py", line 383, in execute

    return Database.Cursor.execute(self, query, params)

sqlite3.OperationalError: 12 values for 6 columns

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File "manage.py", line 15, in <module>

    execute_from_command_line(sys.argv)

  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line

    utility.execute()

  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 375, in execute

    self.fetch_command(subcommand).run_from_argv(self.argv)

  File "/usr/local/lib/python3.7/site-packages/django/core/management/base.py", line 323, in run_from_argv

    self.execute(*args, **cmd_options)

  File "/usr/local/lib/python3.7/site-packages/django/core/management/base.py", line 364, in execute

    output = self.handle(*args, **options)

  File "/usr/local/lib/python3.7/site-packages/django/core/management/base.py", line 83, in wrapped

    res = handle_func(*args, **kwargs)

  File "/usr/local/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 234, in handle

    fake_initial=fake_initial,

  File "/usr/local/lib/python3.7/site-packages/django/db/migrations/executor.py", line 117, in migrate

    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)

  File "/usr/local/lib/python3.7/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards

    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)

  File "/usr/local/lib/python3.7/site-packages/django/db/migrations/executor.py", line 245, in apply_migration

    state = migration.apply(state, schema_editor)

  File "/usr/local/lib/python3.7/site-packages/django/db/migrations/migration.py", line 124, in apply

    operation.database_forwards(self.app_label, schema_editor, old_state, project_state)

  File "/usr/local/lib/python3.7/site-packages/django/db/migrations/operations/special.py", line 190, in database_forwards

    self.code(from_state.apps, schema_editor)

  File "/app/app/migrations/0064_notificationsetting.py", line 54, in forwards

    """)

  File "/usr/local/lib/python3.7/site-packages/django/db/backends/base/schema.py", line 137, in execute

    cursor.execute(sql, params)

  File "/usr/local/lib/python3.7/site-packages/django/db/backends/utils.py", line 67, in execute

    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/usr/local/lib/python3.7/site-packages/django/db/backends/utils.py", line 76, in _execute_with_wrappers

    return executor(sql, params, many, context)

  File "/usr/local/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute

    return self.cursor.execute(sql, params)

  File "/usr/local/lib/python3.7/site-packages/django/db/utils.py", line 89, in __exit__

    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/usr/local/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute

    return self.cursor.execute(sql, params)

  File "/usr/local/lib/python3.7/site-packages/django/db/backends/sqlite3/base.py", line 383, in execute

    return Database.Cursor.execute(self, query, params)

django.db.utils.OperationalError: 12 values for 6 columns
```

I'm also having another probably unrelated issue where docker-compose gives me a warning that the "WEBPACK_LOADER_ENABLED" variable is not set and is defaulting to a blank string, even when I set it to 'True'

kennethjiang, 2022-04-20:
Are you on the `master` branch or the `release` branch?

Erie1, 2022-04-20:
I'm on master

Erie1, 2022-04-20:
just switched to release and the problem seems to be fixed, I should have paid more attention to that

closed 2022-04-20 by Erie1