TheSpaghettiDetective/obico-server #631: Overrides of environment variables through .env don't (always) work

NMe84, 2022-04-20:
**Describe the bug**
I was trying to set up my SMTP server but none of the emails would work. After some troubleshooting with the people on Discord I figured out that the environment variables that I had set up in a `.env` file as suggested by the [readme](https://github.com/TheSpaghettiDetective/TheSpaghettiDetective#configure-email-server-smtp) didn't work.

I had set up my own values in the `.env` file (and removed the `#` in front of them, obviously) but after rebuilding the container and echoing the relevant variables, I got the default values for all of them *except* for some reason `EMAIL_HOST` which does seem to work.

**To Reproduce**
Steps to reproduce the behavior:
1. Check out a clean the currently most recent `release` branch.
2. Create a `.env` file with values for all the relevant email config. Use different values than the default.
3. Bring the containers up and enter a shell with `docker-compose exec web bash`
4. Enter `echo $EMAIL_USE_TLS`. You should see "True" even if you set it to "False".
5. Try `echo $DEFAULT_FROM_EMAIL`. You should see "changeme@example.org" instead of whatever address you entered in the `.env`.

**Hosting environment:**
 - OS: Linux
 - Version: Unsure, whatever Synology uses on their NAS devices
 - Hardware: Synology DS1819+ (quad core Atom CPU C3538 @ 2.10GHz, 32GB RAM)

**Additional context**
When I switched from the `.env` file to just defining the values in `docker-compose.yml` directly, everything worked as intended.

kennethjiang, 2022-04-20:
The new process hadn't been released as new instructions were still on `master` branch. But I can see why this can be confusing. Sorry.

I just released the new process. Can you pull from the `release` branch and try it again?

NMe84, 2022-04-20:
Ah, that actually makes sense. I didn't think to switch branches before reading the documentation, that's as much on me as it is on you, no apologies needed. :smiley:

I can confirm this solves the issue by opening a bash shell in the tasks container and echoing any of the variables, they are properly taken from the `.env` file now.

Sadly I think there may be something wrong with the migration script however. The web container is stuck in a restart loop after the update:

```
web_1     | Operations to perform:
web_1     |   Apply all migrations: account, admin, app, auth, channels_presence, contenttypes, sessions, sites, socialaccount
web_1     | Running migrations:
web_1     |   Applying contenttypes.0001_initial... OK
web_1     |   Applying contenttypes.0002_remove_content_type_name... OK
web_1     |   Applying auth.0001_initial... OK
web_1     |   Applying auth.0002_alter_permission_name_max_length... OK
web_1     |   Applying auth.0003_alter_user_email_max_length... OK
web_1     |   Applying auth.0004_alter_user_username_opts... OK
web_1     |   Applying auth.0005_alter_user_last_login_null... OK
web_1     |   Applying auth.0006_require_contenttypes_0002... OK
web_1     |   Applying auth.0007_alter_validators_add_error_messages... OK
web_1     |   Applying auth.0008_alter_user_username_max_length... OK
web_1     |   Applying auth.0009_alter_user_last_name_max_length... OK
web_1     |   Applying app.0001_initial... OK
web_1     |   Applying account.0001_initial... OK
web_1     |   Applying account.0002_email_max_length... OK
web_1     |   Applying admin.0001_initial... OK
web_1     |   Applying admin.0002_logentry_remove_auto_add... OK
web_1     |   Applying admin.0003_logentry_add_action_flag_choices... OK
web_1     |   Applying app.0002_seed_data_2247... OK
web_1     |   Applying app.0003_auto_20190202_1354... OK
web_1     |   Applying app.0004_auto_20190210_1254... OK
web_1     |   Applying app.0005_auto_20190219_2337... OK
web_1     |   Applying app.0006_bed_heater_off_default_false... OK
web_1     |   Applying app.0007_add_phone_num_to_user... OK
web_1     |   Applying app.0008_create_printer_preidction... OK
web_1     |   Applying app.0009_add_detective_sensitivity_to_printer... OK
web_1     |   Applying app.0010_auto_20190303_2125... OK
web_1     |   Applying app.0011_print_prediction_json_url... OK
web_1     |   Applying app.0012_auto_20190323_2103... OK
web_1     |   Applying app.0013_add_alert_fields_to_print... OK
web_1     |   Applying app.0014_remove_print_fields_from_printer... OK
web_1     |   Applying app.0015_print_alert_muted_at... OK
web_1     |   Applying app.0016_add_index_json_url_to_print... OK
web_1     |   Applying app.0017_user_pushbullet_access_token... OK
web_1     |   Applying app.0018_add_alert_overwrite_to_print... OK
web_1     |   Applying app.0019_usercredit... OK
web_1     |   Applying app.0020_user_telegram_chat_id... OK
web_1     |   Applying app.0021_make_printer_id_nullable_in_print... OK
web_1     |   Applying app.0022_auto_20190821_2145... OK
web_1     |   Applying app.0023_add_consented_at_to_user... OK
web_1     |   Applying app.0024_auto_20191107_1903... OK
web_1     |   Applying app.0025_auto_20191107_2013... OK
web_1     |   Applying app.0026_auto_20191110_1315... OK
web_1     |   Applying app.0027_add_account_status_to_user... OK
web_1     |   Applying app.0028_sharedresource... OK
web_1     |   Applying app.0029_auto_20191212_2101... OK
web_1     |   Applying app.0030_auto_20191213_0510... OK
web_1     |   Applying app.0031_auto_20191215_2332... OK
web_1     |   Applying app.0032_gcodefile... OK
web_1     |   Applying app.0033_user_slack_access_token... OK
web_1     |   Applying app.0034_auto_20200311_1805... OK
web_1     |   Applying app.0035_auto_20200522_0843... OK
web_1     |   Applying app.0036_auto_20200612_2224... OK
web_1     |   Applying app.0037_auto_20200629_0127... OK
web_1     |   Applying app.0038_printshotfeedback_persisted_at... OK
web_1     |   Applying app.0039_auto_20200728_2028... OK
web_1     |   Applying app.0040_auto_20201123_0452... OK
web_1     |   Applying app.0041_auto_20201126_0331... OK
web_1     |   Applying app.0042_auto_20201202_1937... OK
web_1     |   Applying app.0043_auto_20201212_0327... OK
web_1     |   Applying app.0044_auto_20210105_0438... OK
[... 377 lines cut ...]
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/db/migrations/executor.py", line 245, in apply_migration
web_1     |     state = migration.apply(state, schema_editor)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/db/migrations/migration.py", line 124, in apply
web_1     |     operation.database_forwards(self.app_label, schema_editor, old_state, project_state)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/db/migrations/operations/special.py", line 190, in database_forwards
web_1     |     self.code(from_state.apps, schema_editor)
web_1     |   File "/app/app/migrations/0064_notificationsetting.py", line 54, in forwards
web_1     |     """)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/db/backends/base/schema.py", line 137, in execute
web_1     |     cursor.execute(sql, params)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/db/backends/utils.py", line 67, in execute
web_1     |     return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
[... cut ...]
web_1     |     raise dj_exc_value.with_traceback(traceback) from exc_value
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
web_1     |     return self.cursor.execute(sql, params)
web_1     |   File "/usr/local/lib/python3.7/site-packages/django/db/backends/sqlite3/base.py", line 383, in execute
web_1     |     return Database.Cursor.execute(self, query, params)
web_1     | django.db.utils.OperationalError: 12 values for 6 columns
web_1     |   Applying app.0064_notificationsetting...
```

kennethjiang, 2022-04-20:
@NMe84 Pushed a fix. Please pull it then run `docker-compose up --build -d`.

If it still doesn't work, please run `rm backend/db.sqlite3 && docker-compose up --build -d`. Please note this will erase your database but I assume you have a new installation so it shouldn't matter.

NMe84, 2022-04-20:
That fixed it, though even without removing the database I had to start fresh anyway. Not that it matters because as you assumed it was a fresh install anyway, so I didn't lose anything. Thanks!

closed 2022-04-21 by kennethjiang