TheSpaghettiDetective/obico-server #636: [BUG] Update to latest release removes existing database

querex007, 2022-05-04:
Seems there were 2 new releases on the "release branch" as of 27 april.
I upgraded my private spaghetti detective, but seems the database is rebuild / cleared.

I followed the instructions described in the readme (from the release branch).
- Created the .env (as custom settings are no longer in the docker-compose.yaml file).
- Followed the upgrade instructions "git checkout release && git pull && docker-compose up --build -d"

After upgrade, docker containers are back up & running, but seems all user data is cleared after the update.
Guess some instructions are missing or their is a bug in the upgrade script?
I always make a proxmox CT backup before I start an upgrade, so no blocking issue for me :)
Many thanks for the great project!

raymondh2, 2022-05-04:
The data is not cleared. The directory it is stored in is different. Move the database file and static build folder to the backend folder fomr the web folder

kennethjiang, 2022-05-05:
@querex007 You should get your database back once you did what @raymondh2  said. Basically `docker-compose stop && cp backend/db.sqlite3 backend/db.sqlite3.back && mv web/db.sqlite3 backend/db.sqlite3 && docker-compose up -d".

Sorry for this bug!