TheSpaghettiDetective/obico-server #643: Documentation mismatch on .env file

MallocArray, 2022-05-15:
**Describe the bug**
On this page:
https://www.obico.io/docs/server-guides/configure/#email-smtp
the example says to find these variables and set them to the correct values, but the leading # is still present, which typically indicates the line is commented out and will not take effect.  

On another page
https://www.obico.io/docs/server-guides/email_guide/
The leading # is not present, which I believe is the correct method, but wanted to double check.  This code block is also indented with 2 spaces, which I don't think is necessary/supported

It also mentions modifying docker-compose.yaml which isn't necessary anymore since the .env file is being used.  I'm unsure if `docker-compose up --build -d` is still needed to accept the new values from the .env file or not.

**To Reproduce**
Steps to reproduce the behavior:
1. ...
2. ...
3. ...

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]
 - Hardware [e.g., PC, a Jetson Nano]

**Additional context**
Add any other context about the problem here.

kennethjiang, 2022-05-16:
Than you @MallocArray for reporting these problems in the doc. I pushed https://github.com/TheSpaghettiDetective/obico-server/commit/e3e0c80f0a23923d5a2e8aacaf5f95ebfcf7f355

Not entirely sure about the deal around `"` in `.env`. It worked for me on Mac. I'm suspecting it's related to the docker version/platform...? Can you do a bit more research on this topic? I don't want to change it so that it fix a problem for some platforms but causes a problem for others.

MallocArray, 2022-05-16:
I know that changing `EMAIL_HOST=smtp.office365.com` to `EMAIL_HOST="smtp.office365.com"` stopped it from being able to send in my Oracle Cloud Instance running Ubuntu 20.04.  I'll have to get the exact Docker version that I'm running.

I can test it again as I was attempting several things when I started narrowing down to just that.

The comments also came through in my environment variables inside of the container which I think adds to the issue.  If you have comments in the .env lines, can you check what value you see in the container?
So if your .env has `EMAIL_HOST="smtp.office365.com"  # Such as "smtp.gmail.com"` what do you see when you run
```
sudo docker exec obico-server_web_1 env
```

In my output, I was seeing
```
EMAIL_HOST="smtp.office365.com"  # Such as "smtp.gmail.com"
```
which means the actual environment variable inside of the container had the full string just as it was, including the comment, while I would have expected that to be stripped out.

kennethjiang, 2022-05-16:
I think [this](https://github.com/docker/compose/issues/8388) is relevant.

MallocArray, 2022-05-16:
Are you using straight docker then?  From how I read that link, it sounds like there is a difference in how docker and docker-compose treat the variable.  Or are you using a newer version of docker-compose maybe?

This is what I get when using the latest Ubuntu 20.04 template from Oracle Cloud and install docker-compose with apt

```
sudo apt install docker-compose jq zip -y

$ docker-compose --version
docker-compose version 1.25.0, build unknown
$ docker --version
Docker version 20.10.12, build 20.10.12-0ubuntu2~20.04.1
```

kennethjiang, 2022-05-16:
https://github.com/TheSpaghettiDetective/obico-server/commit/c079c45afc3c11258751ea1f969798f2b64c6e1c

kennethjiang, 2022-05-16:
It looks like the newer version of docker-compose handles quotes. But it should be safe to remove quotes based on the official docs.

closed 2022-05-16 by kennethjiang

MallocArray, 2022-05-16:
ok, thanks.  From this link, it looks like I need to use a different package name to install the latest version of compose, which I'll test out tomorrow
https://docs.docker.com/compose/install/
`sudo apt-get install docker-compose-plugin`