TheSpaghettiDetective/obico-server #662: [BUG] ACCOUNT_ALLOW_SIGN_UP does not disable signup, only hides button

marrelmaker, 2022-06-27:
**Describe the bug**
When setting ACCOUNT_ALLOW_SIGN_UP=False in the .env-file, the sign up button vanishes in the UI (which is to be expected). However, if you visit https://<your-obico.com>/accounts/signup/ by entering the URL manually in the browser or from the button below a shared live feed (while logged out), the sign up form still appears and you are successfully able to create an account.
This should not be possible if sign ups are disabled, at least to my understanding.

**To Reproduce**
Steps to reproduce the behavior:
1. set ACCOUNT_ALLOW_SIGN_UP=False and restart the docker stack
2. visit https://<your-obico.com>/accounts/signup/ 
3. create an account without issues, even though sign up is disabled

**Screenshots**
-

**Hosting environment (please complete the following information):**
 - OS: Debian 11 / docker-compose 2.6 / docker 20.10.17
 - Version: release branch commit b469cdd (latest at this time)
 - Hardware: KVM based virtual server

**Additional context**
-

kennethjiang, 2022-06-29:
Good catch.

phoenixflare, 2023-01-07:
Still not fixed. For now anyone running this behind the nginx proxypass can just add the following to disable it:
`  location /accounts/signup/ {
    return 301 http://$host/accounts/login/;
  }
`

kennethjiang, 2023-01-10:
Help wanted! A PR for a more permanent fix will be really appreciated!

marrelmaker, 2023-10-10:
It's been quite a while, but I just disabled my workaround to test this with the latest release and it's working now as inteded. Thank you @phoenixflare !

closed 2023-10-10 by marrelmaker