TheSpaghettiDetective/obico-server #667: [Feature] Support basic auth in iOS app

com6056, 2022-07-18:
**Is your feature request related to a problem? Please describe.**
Obico iOS app doesn't support Obico self-hosted servers with basic authentication in front of them.

**Describe the solution you'd like**
I'd love a way to input basic authentication credentials in the iOS app when logging into a self-hosted Obico server.
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
I've considered disabling my authentication layer that I have in front of Obico, but I'd be much more comfortable being able to keep it.

**Additional context**
Attempting to use basic authentication credentials in the URL (`https://username:password@example.com`) breaks the app and all I get is a blank white screen when opening it afterwards until I delete it and re-install it.

kennethjiang, 2022-07-19:
Currently, the app would  ask you to login again only when you haven't logged in for 3 months. Isn't this a good enough alternative?

com6056, 2022-07-19:
I'd like to avoid exposing Obico to the open internet with only it's built-in authentication, I'd be much more comfortable with keeping my own authentication layer in front of it. If the app could just allow putting in basic auth creds, storing them, and passing those along in the requests, that should solve it.

kennethjiang, 2022-07-19:
Do you mean the basic auth *in addition to*, not "instead of" the built-in auth?

com6056, 2022-07-19:
Yep, just want the ability to pass basic auth creds in each request sent by the iOS client 🙌 No need to remove anything