TheSpaghettiDetective/obico-server #668: [Feature] Support disabling the built-in authentication

com6056, 2022-07-18:
**Is your feature request related to a problem? Please describe.**
Nope, I just roll my own authentication layer and would love to disable Obico's built-in authentication so I don't have to login twice.

**Describe the solution you'd like**
Being able to disable Obico's built-in authentication.

kennethjiang, 2022-07-18:
Do you think you can just implement this in your own fork, since it sounds like this is something very specific to your own setup.

com6056, 2022-07-19:
This is a pretty common feature asked for in self-hosted projects, even OctoPrint supports this: https://docs.octoprint.org/en/master/features/accesscontrol.html#autologin

I think just being able to do an auto-login like OctoPrint would be enough, no need to completely disable the auth system entirely. Happy to dig into it a bit when I have some free time though!

cp2004, 2022-07-19:
> This is a pretty common feature asked for in self-hosted projects, even OctoPrint supports this: https://docs.octoprint.org/en/master/features/accesscontrol.html#autologin

Autologin shouldn't be confused with actually disabling authentication, it is a different thing.

Allowing people to disable authentication is not something we would *like* to have in OctoPrint, it's just the way that forced login had to be implemented due to too many people not setting up authentication properly and exposing everything to the internet stupidly. It is then very difficult to completely remove these options so 'workarounds' such as autologin with `0.0.0.0/0` exist because people come kicking and screaming on the doorstep insulting every living thing in sight if you upset them.