TheSpaghettiDetective/obico-server #674: [BUG] stuck on 0.1fps for video streaming

3DCoded, 2022-08-02:
**Describe the bug**
In my OctoPrint interface, I setup a streaming server that goes at around 60fps. When I connect it to my self-hosted Obico server on the same machine, it only updates the webcam when I reload, and only a few times a minute.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Raspbian
2. Install OctoPrint
3. Install Obico
4. Check Obico's video feed vs OctoPrint's

**Hosting environment (please complete the following information):**
 - OS: Raspbian Bullseye
 - Version: 11
 - Hardware: Raspberry Pi 400

kennethjiang, 2022-08-04:
This is most likely due to an environment issue. For instance: https://www.obico.io/docs/server-guides/faqs/#why-doesnt-the-premium-streaming-work

I will close the issue for now. Please come to our discord server obico.io/discord/. You will usually get help a lot more quickly this way.

3DCoded, 2022-08-04:
Unfortunately, I do not have access to Discord. I am running OctoPi in a Docker container

kennethjiang, 2022-08-04:
Discord is free and is quite easy to join.

In any case, the issue you ran into is probably https://www.obico.io/docs/user-guides/webcam-stream-stuck-at-1-10-fps/#2-is-octoprint-running-on-a-raspberry-pi

3DCoded, 2022-08-04:
I already tried everything on that page to no avail.

kennethjiang, 2022-08-05:
It's not a bug. The video streaming requires using octopi image. https://www.obico.io/docs/user-guides/webcam-stream-stuck-at-1-10-fps/#2-is-octoprint-running-on-a-raspberry-pi

3DCoded, 2022-08-05:
That is what I am using.

3DCoded, 2022-08-05:
I am using the octopi image in docker.

cp2004, 2022-08-05:
Check the difference between Octo*Print* and Octo*Pi* here:
https://community.octoprint.org/t/what-is-the-difference-between-octoprint-and-octopi-are-they-the-same-thing/185

You can't run Octo**Pi** in docker - it is an operating system for a Raspberry Pi. You are using Octo**Print** in a container, and the docs link above says that this does not work for premium streaming.

3DCoded, 2022-08-05:
When I go to Obico Plugin settings in OctoPrint, it says streaming mode is Premium. This seems still a bug and should be fixed for OctoPrint.

kennethjiang, 2022-08-05:
This is a fair point. But it should be a bug in the Obico plugin, not OctoPrint.

I changed the title and reopened this issue. Can you post a screenshot for it?

3DCoded, 2022-11-01:
[image]

blockofdynamite, 2023-07-31:
FWIW, I have also just run into this bug.

Fresh obico server, fresh octopi image on an rpi3 with a usb webcam. Server and Webcam connection show as green in the plugin status. Webcam works fine in octoprint on both the always- and never-compatibility settings. However, obico server only shows a snapshot.