TheSpaghettiDetective/obico-server #677: [BUG] Cannot make API calls without authentication

davidloenborg, 2022-08-09:
I want to make API calls to obico from home assistant.

**Describe the bug**
When I browse the link xx.xx.xx/api/v1/printers/1 in a edge, I can see the results just fine. But in a guest mode browser, or from postman with basic auth, I cannot get passed "Authentication credentials was not provided"

Kenneth suggested that I tried tunnel mode, but I was not able to access obico API that way through, only octoprint.

Steps to reproduce the behavior:
Try to access the obico API from a browser where you are logged in, and then from incognito.

**Screenshots**
Here are login from webbrowser
[image]

And here from postman
[image]

**Hosting environment (please complete the following information):**
Obico server runs in a VM with 4 cores and 4GB memory
Octoprint runs on raspberry pi 4B with 4GB memory, and 64GB SD card
Client devices is Windows 11 on latest release

**Additional context**
API keys would be nice.
If there are questions, please let me know.

johngoadams, 2022-09-13:
Was there any update on this? I got to the same point as you trying to use the obico cloud APIs

kennethjiang, 2022-09-13:
@johngoadams Did you try to create a  basic auth username/password pair using the authorization page? https://obico.io/docs/user-guides/tunnel-app-integration/ 

I was unable to reproduce this bug on my local dev env. But it may be a bug that only happens in prod. I'd like to know how many users are impacted by this bug.

johngoadams, 2022-09-14:
I might have been using the API documentiaon incorrectly. I was not trying to tunnel through but gather information from the Obico cloud. It seems to work within a browser that I am logged in on but using a private window (sans cookies) I get prompted for authentication. Not sure how to pass this to the API.