TheSpaghettiDetective/obico-server #678: [Feature] Additional Email Methods for Self-Hosted Servers

ajn142, 2022-08-17:
**Is your feature request related to a problem? Please describe.**
I don’t currently maintain an SMTP server, and am already paying for an e-mail service (through Microsoft’s 365 offering, but I’ve used Google’s GSuite/Workspaces as well). Most of these services are moving away from SMTP to API-based email methods protected by better authentication methods, including MFA.

**Describe the solution you'd like**
I’d like to see support for alternative methods to send email beyond SMTP (e.g. the GMail API https://developers.google.com/gmail/api/reference/rest or Microsoft’s Graph API https://docs.microsoft.com/en-us/graph/api/resources/mail-api-overview?view=graph-rest-1.0) that support better authentication methods added. This would be useful for the self-hosted servers (my case) and may also provide a leg up on integrating these services in the cloud offering in the future.

**Describe alternatives you've considered**
I’ve considered implementing an SMTP server for my domain, but the added overhead of managing it for this single application isn’t worthwhile. My short term solution is likely to be using a separate free email account for notifications, but given I’m already paying for a service for email from my domain, using a generic account isn’t desirable long term.

**Additional context**
Add any other context or screenshots about the feature request here.

kennethjiang, 2022-08-17:
I like this idea! A quick google search shows this is possible, although not without effort: https://www.codersarts.com/post/how-to-integrate-gmail-api-with-django

We will need to rely on the community contribution for this feature as the core team has our hands full now.

charredchar, 2023-02-11:
There is one more option I would like to purpose, use a ready-made SMTP Relay  Docker container.

I personally run an SMTP Relay on my domain linked to a Google Gmail specifically to deal with multiple services I have in-house so they can all go through a single point instead of needing to configure each one individually. I would think it might be a lot easier to find a docker image already made that can be utilized and deployed during install with minimal effort instead of trying to do the same within Obico.

kennethjiang, 2023-02-15:
> There is one more option I would like to propose, use a ready-made SMTP Relay Docker container.
> 
> I personally run an SMTP Relay on my domain linked to a Google Gmail specifically to deal with multiple services I have in-house so they can all go through a single point instead of needing to configure each one individually. I would think it might be a lot easier to find a docker image already made that can be utilized and deployed during install with minimal effort instead of trying to do the same within Obico.

Love your idea! Can you experiment to see if your idea works?

ajn142, 2023-12-09:
> There is one more option I would like to propose, use a ready-made SMTP Relay Docker container.
>
> I personally run an SMTP Relay on my domain linked to a Google Gmail specifically to deal with [...]

I just wanted to come back and say this is the route I’ve gone. I haven’t tested with Obico yet though.

https://github.com/ajn142/smtp2graph