TheSpaghettiDetective/obico-server #691: [BUG] Cannot login to self-hosted server

dannyglover, 2022-09-01:
**Describe the bug**

I have set up my Obico self hosted server following the official guide provided by Obico. I installed Docker, installed iTunes (for Bonjour) and everything seemed ok. I could access the local Obico login page and believed I was set to continue.

However, no matter what I do, I can't seem to be able to login to my local Obico instance.
I searched the issues list and tried googling the issue, but cannot seem to find any information on this particular issue.

I'm happy to provide any additional info needed, including logs. Just point me towards the location of the data you require and I'll get it here asap.

Thank you.

**To Reproduce**
Steps to reproduce the behavior:

1) Follow the self-install guide from Obico on Windows via Docker.
2) Install iTunes as recommended for Bonjour.
3) Get hostname (in my case AcerNitro).
4) Open the site via: http://acernitro.local:3334/accounts/login/  - Same issue exists via localhost address.
5) Enter your account credentials from Obico - I have verified my login works correctly on the Obico website.
6) Click login and it throws the following error on the webpage: "Wrong email or password".

**Hosting environment (please complete the following information):**
 - Windows
 - 21H2
 - Ryzen based laptop (R7 4800H, 16GB DDR4, GTX 1650TI, 500GB NVME)

**Additional context**

Here is some of the log ouput from Docker:

```
web_1 | daphne.server INFO     HTTP/2 support not enabled (install the http2 and tls Twisted extras)
web_1 | daphne.server INFO     Configuring endpoint tcp:port=3334:interface=0.0.0.0
web_1 | daphne.server INFO     Listening on TCP address 0.0.0.0:3334
web_1 | django.channels.server INFO     HTTP GET /accounts/login/ 200 [1.13, 172.18.0.1:42354]
web_1 | django.channels.server INFO     HTTP GET /static/vendor/css/sweetalert2.min.fbf274a0f38a.css 200 [0.02, 172.18.0.1:42354]
web_1 | django.channels.server INFO     HTTP GET /static/frontend/js/chunk-vendors.b9753c301ba8.js 200 [0.02, 172.18.0.1:42358]
web_1 | django.channels.server INFO     HTTP GET /static/vendor/js/filesize.min.ef678061e185.js 200 [0.03, 172.18.0.1:42356]
web_1 | django.channels.server INFO     HTTP GET /static/hijack/hijack-styles.a7cafac63295.css 200 [0.03, 172.18.0.1:42360]
web_1 | django.channels.server INFO     HTTP GET /static/js/static-color-scheme-controller.0755f55750b9.js 200 [0.03, 172.18.0.1:42362]
web_1 | django.channels.server INFO     HTTP GET /static/frontend/js/styles.10116f0b950b.js 200 [0.03, 172.18.0.1:42364]
web_1 | django.channels.server INFO     HTTP GET /static/frontend/css/styles.9fddba8f16eb.css 200 [0.04, 172.18.0.1:42354]
web_1 | django.channels.server INFO     HTTP GET /static/frontend/css/chunk-vendors.03e0fb6dbd75.css 200 [0.04, 172.18.0.1:42356]
web_1 | django.channels.server INFO     HTTP GET /static/vendor/js/polyfill.min.51c3377e1cd4.js 200 [0.04, 172.18.0.1:42360]
web_1 | django.channels.server INFO     HTTP GET /static/vendor/js/jquery-3.4.1.min.a6b6350ee94a.js 200 [0.04, 172.18.0.1:42358]
web_1 | django.channels.server INFO     HTTP GET /static/vendor/js/popper.min.855075c14fed.js 200 [0.04, 172.18.0.1:42362]
web_1 | django.channels.server INFO     HTTP GET /static/vendor/js/bootstrap.min.7451e07cd607.js 200 [0.04, 172.18.0.1:42364]
web_1 | django.channels.server INFO     HTTP GET /static/vendor/js/clipboard.min.1ebaf9d368db.js 200 [0.03, 172.18.0.1:42354]
web_1 | django.channels.server INFO     HTTP GET /static/vendor/js/bootstrap-input-spinner.min.8e7342aefdfb.js 200 [0.03, 172.18.0.1:42356]
web_1 | django.channels.server INFO     HTTP GET /static/vendor/js/download.min.264a62617263.js 200 [0.02, 172.18.0.1:42360]
web_1 | django.channels.server INFO     HTTP GET /static/js/color-scheme.js 200 [0.02, 172.18.0.1:42358]
web_1 | django.channels.server INFO     HTTP GET /static/js/utils.js 200 [0.01, 172.18.0.1:42358]
web_1 | django.channels.server INFO     HTTP GET /static/img/loader/loader_dark-scheme.f5acdf2cb72a.svg 200 [0.01, 172.18.0.1:42366]
web_1 | django.channels.server INFO     HTTP GET /static/img/favicon.a47b1972c1bb.png 200 [0.01, 172.18.0.1:42358]
web_1 | django.channels.server INFO     HTTP POST /accounts/login/ 200 [0.10, 172.18.0.1:42358]
web_1 | django.channels.server INFO     HTTP POST /accounts/login/ 200 [0.08, 172.18.0.1:42358]
web_1 | django.channels.server INFO     HTTP GET /static/js/color-scheme.js 304 [0.00, 172.18.0.1:42358]
web_1 | django.channels.server INFO     HTTP GET /static/js/utils.js 304 [0.00, 172.18.0.1:42358]
```

Orafilynie, 2022-09-01:
Are you using your credentials from your real obico account (Online) to try login ?

dannyglover, 2022-09-01:
> Are you using your credentials from your real obico account (Online) to try login ?

I am indeed, yeah. Is that incorrect?

Orafilynie, 2022-09-01:
> > Are you using your credentials from your real obico account (Online) to try login ?
> 
> I am indeed, yeah. Is that incorrect?

Then you are not following the guide

Orafilynie, 2022-09-01:
You need to create an account with the django interface

dannyglover, 2022-09-01:
> > > Are you using your credentials from your real obico account (Online) to try login ?
> > 
> > 
> > I am indeed, yeah. Is that incorrect?
> 
> Then you are not following the guide

Indeed, I totally missed that section. 
Apologies, I'll close this.

Thank you for pointing that out!

closed 2022-09-01 by dannyglover

Orafilynie, 2022-09-01:
👍

kennethjiang, 2022-09-02:
Thank you @Orafilynie for helping out fellow Obico community members!