TheSpaghettiDetective/obico-server #695: Using hitend

rrauenza, 2022-09-06:


rrauenza, 2022-09-07:
I was thinking start collecting the temperature data with your user base submitted images of failures.   Then you've got temperature data classified with success and failures...

I'm wondering if the FFT changes with a blob... lower frequency?  I think the magnitude wouldn't change due to the PID.  But corrective actions may be less frequent with the extra blob insulation?

At the very least, just start collecting it... so a year from now you have it to poke at.

Maybe simulate it by insulating the hot end and seeing how the PID reacts?  See if there is signal in that noise worth pursuing?  Obviously using real data later if you think it is detectable...

kennethjiang, 2022-09-08:
I like your idea of simulating the problem by insulating the hot end and observe if the data come out more or less in line as what the theory predicts. Are you interested in giving it a shot?

rrauenza, 2022-09-12:
No, I don't have the time these days.  My thought was to suggest collecting the data from deployments along with images so that it could be analyzed someday.

rrauenza, 2022-10-04:
Prusa is putting blob detection in their firmware… with some data analysis in the blog on temperature patterns.  You might be interested in this:

https://blog.prusa3d.com/mk3s-3-12-beta-firmware-new-thermal-model-protection-and-blob-detection_71230/

kennethjiang, 2022-10-05:
Thank you for the link! Very interesting!