TheSpaghettiDetective/obico-server #699: [Feature] Filament Tracking

ITredneck, 2022-09-17:
The only feature I am missing to not need to use octofarm is to be able to track filament that I use among my printers. Currently I am re-weighing them every time I use them because it does not track across my printers. It would be nice to be able to set up filament tracking in Obico as it works across all my printers and I wouldn't need to setup an octofarm server just for that feature anymore..

kennethjiang, 2022-09-17:
Thank you for submitting this feature request. This is awesome because we recently have started thinking about putting this feature on the roadmap.

@neilhailey You have quite a few printers. Maybe you can set up octofarm and study how filament tracking works in Octofarm?

ITredneck, 2022-09-17:
Really the app to look at is how fillament manager on octoprint works. Octofarm just has the app and the apps on the printers repoint their database back to octofarm database instead of using it locally.

kennethjiang, 2022-09-17:
[This plugin](https://plugins.octoprint.org/plugins/filamentmanager/)? If so, maybe Obico plugin can just directly talk to this plugin to fetch the info, instead of trying to reinvent the wheel?

ITredneck, 2022-09-18:
Yes, that plugin. From my understanding its all just going back to a database. I know this because the setup for octofarm, is to migrate the database to the octofarm server and then point the plugin to the octofarm server for its data. So you could do a sync job between the databases of some kind as well as pull your data from there.

kennethjiang, 2022-09-21:
Gotcha! Thank you so much for the info!