TheSpaghettiDetective/obico-server #706: [BUG] Self-hosted Docker containers send pushover notifications with broken images

rcarmo, 2022-10-06:
**Note**
Most of the times, an issue you may potentially run into is not a bug. Instead, it is related to configurations, such as firewall rules, incompatible docker version, reverse proxy setups, etc.

If you are not sure, please don't open a bug report here. Instead, you can get the issue solved much faster by seeking help in our support forum: https://obico.io/discord.

Proceed to opening a bug report only if you are sure it is a defect in the code given a correct configuration.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Install self-hosted server and verify it can access camera (server is on the same LAN as Klipper)
2. Set up Pushover User and App (appkey in docker-compose ENV, user key in web UI)
3. Start a print

**Screenshots**
N/A

**Hosting environment (please complete the following information):**
 - OS: Ubuntu (Docker), running the latest version as of October 5th, on an Intel PC

**Additional context**
I get all notifications just fine, but they all show up with a broken image. My guess, after reading through the source code for the plugin, is that it is not being handed a valid image file to POST to the Pushover server (I did not find how the plugin was loaded and what it was handed as a payload to send).

I have my own notification scheme that does that successfully, so I can debug this if I get some pointers - but my guess is that the self-hosting inside Docker breaks some assumptions that were made when the code was written.

kennethjiang, 2022-10-07:
Yes if you can help us debug it, it'll definitely speed up our work to fix it.

The relevant code is https://github.com/TheSpaghettiDetective/obico-server/blob/release/backend/notifications/plugins/pushover/__init__.py#L105

You can also link your printer to the Obico cloud (app.obico.io) to see if image attachment works there. If so, we can narrow down the problem to the very limited places where the cloud is different from the self-hosted server.

tautomer, 2022-10-24:
Not sure if I'm having the same issue, but it does sounds like related. (I will open a new issue, it this is actually something else.)

My server detects failures, but the video stream is a broken icon. The timelapse is broken as well.

[image]

[image]

Stream in Octoprint works fine. SSH into the container (tested `spaghetti_detective-web`), `wget` downloads the video just fine.

Before the server upgrade (installed back in February), video was fine, but obviously it wasn't compatible with the new Octoprint plugin.

kennethjiang, 2022-10-27:
@tautomer What you said sounds more like a configuration issue than a bug. The quickest way to get help is to come to our discord server: obico.io/discord/