TheSpaghettiDetective/obico-server #711: [BUG] Image attachment in Discord stopped working

kennethjiang, 2022-11-06:
It looks like some subtle change in Discord API that has broke the image attachment function.

Steps to reproduce:

```
from discord_webhook import DiscordWebhook, DiscordEmbed
webhook = DiscordWebhook(url='your discord webhook url')
embed = DiscordEmbed(title='Your Title', description='Lorem ipsum dolor sit', color='03b2f8')
embed.set_image(url='https://app-stg.obico.io/ent/object_store/?t=0/tsd-pics/snapshots/244/1667691350.71139_rotated.jpg&d=48dd0bffee71c77327a9d36374c1332e&ts=1667691350.7235715')
response = webhook.execute()
```

The message sent won't have the image.

If
```
embed.set_image(url='https://app-stg.obico.io/ent/object_store/?t=0/tsd-pics/snapshots/244/1667691350.71139_rotated.jpg&d=48dd0bffee71c77327a9d36374c1332e&ts=1667691350.7235715')
```
is changed to a simpler image url, such as
```
embed.set_image(url='https://obico.io/img/favicon.png')
```
the message will have the image correctly attached.

My best guess is some recent change in Discord API has broken its previous behavior. This looks more like a bug in Discord API, so we could just wait it out until Discord fixes it. But it's also possible that it's a "feature" by their design.

Help wanted as we don't have time to dig much deeper atm.

pilonull, 2023-05-21:
Would love to have image snippets in my obico discord notifications, that would be awesome.

lustsazeus-lab, 2026-02-24:
Given Discord likely tightened URL validation, an effective workaround is to host image links as clean, extension-based URLs (or CDN-signed URLs without query-heavy params) before passing to `embed.set_image`.

If you can share one failing URL and one transformed clean URL that works, maintainers can codify a safe normalization rule (or fallback to attachment upload) in one patch.