TheSpaghettiDetective/obico-server #716: [Feature] Notification include link to stream

cr45hmurphy, 2022-11-13:
**Is your feature request related to a problem? Please describe.**
Not related to any current issue or problem.

**Describe the solution you'd like**
It would be really cool if notifications included a link to the stream. I use the Discord webhook, and it would be cool if the other members could easily click on the notification and see what I'm printing.

**Describe alternatives you've considered**
None that I know of at this moment.

**Additional context**
None at the moment.

kennethjiang, 2022-11-14:
@cr45hmurphy Obico support many different kinds of notificaitons. Do you mean just Discord, or all notification channels?

cr45hmurphy, 2022-11-14:
> @cr45hmurphy Obico support many different kinds of notificaitons. Do you mean just Discord, or all notification channels?

In my case, Discord. But, from what I can tell, none of the other notifications send a stream link.

Regarding Discord, I'm specifically thinking that a stream link be sent out with a "print start" notification.

kennethjiang, 2022-11-14:
@cr45hmurphy We have a bug open for discord: https://github.com/TheSpaghettiDetective/obico-server/issues/711

Can you test it for other channels to see if you experience the same issue?

cr45hmurphy, 2022-11-14:
> @cr45hmurphy We have a bug open for discord: #711
> 
> Can you test it for other channels to see if you experience the same issue?

My Discord notifications haven't ever actually sent images. It's always just been a text notification.

kennethjiang, 2022-11-14:
This sounds like just https://github.com/TheSpaghettiDetective/obico-server/issues/711. Let me know if you still think you are experiencing a different problem. Otherwise I'll close it.

cr45hmurphy, 2022-11-14:
Does the image normally link to the printing stream? If not, then this is a different request.

kennethjiang, 2022-11-15:
I know what you mean now.

Is the reason why you want the share link to be included is so that you can easily grab it to send it to other people?

cr45hmurphy, 2022-11-15:
> I know what you mean now.
> 
> 
> 
> Is the reason why you want the share link to be included is so that you can easily grab it to send it to other people? 

Currently, I utilize the Discord webhook to send a notification to a text channel when a print job starts. With the ability to send notifications to collaborative communication platforms (Discord, Slack, etc.) there is the inherent idea of inviting others to view the stream. Otherwise, the notifications are simply there for the owner of the print, and that seems somewhat unnecessary.

The hope is that when a print starts and an associated notification gets posted on one of those platforms (Discord, in my case), other users would be able to simply click on the notification and be taken to the live feed of the print, so that they can watch along if they want to. 

I realize that the owner (ie. Me) can simply share the live stream URL as well. However, that can change if live stream is ever turned off/on. It just seems that adding that capability to the actual notification would make sharing the feed a little bit easier.

kennethjiang, 2022-11-15:
Do you mean Obico should create a shared link if one doesn't exist yet? I'm afraid this will surprise many users.

kennethjiang, 2022-11-16:
An alternative design is to include the shared link only if it has been manually created by the account owner. In this case, it may still come as an unexpected surprise. This shared link is supposed to be confidential and only shared manually by the account owner.

cr45hmurphy, 2022-11-16:
> An alternative design is to include the shared link only if it has been manually created by the account owner. In this case, it may still come as an unexpected surprise. This shared link is supposed to be confidential and only shared manually by the account owner.

What about a way for the owner to select an option to allow for the sharing of the feed link? Like, a way that the user could turn off the sharing of the feed URL if they wanted to.

kennethjiang, 2022-11-16:
Good point. This option can be a checkbox in the sharing dialog. Good call.

cr45hmurphy, 2023-04-15:
Is there a way to have this moved to the Octopi repo instead? I just realized I put it in the wrong repo.

kennethjiang, 2023-04-18:
I'm confused. This seems like a feature request for Obico. Why do you want it to be in Octopi? In any case, if you want it in the octopi repo, please contact the octopi repo owner.

cr45hmurphy, 2023-04-18:
Sorry, I meant the Obico Octopi repo. I didn't mean to have this placed in the Obico Server repo.

cr45hmurphy, 2023-04-18:
Obico Octoprint, rather (https://github.com/TheSpaghettiDetective/OctoPrint-Obico)

kennethjiang, 2023-04-18:
This is a server-side feature. So this is the right repo for it.

cr45hmurphy, 2023-04-18:
Okay. I thought this repo was specifically for setting up your own Obico server? Does it also affect features on the cloud-hosted option as well?

kennethjiang, 2023-04-18:
Obico cloud and self-hosted Obico servers are essentially the same code.

cr45hmurphy, 2023-04-18:
Oh okay! Wanted to ensure I wasn't putting this in the wrong repo. Thank you for the clarification.