TheSpaghettiDetective/obico-server #728: USB camera rotation / settings

OwendB1, 2022-12-08:
**Is your feature request related to a problem? Please describe.**
I have been trying for 2 days to use MJPEG streamer and/or FFMPEG to try and rotate my camera 180 degrees because physically rotating the camera is not possible. I'm trying to accomplish this on a USB camera (C270) with FLuidd and Mainsail.

**Describe the solution you'd like**
It would be ideal if Obico allowed for a postprocessing setting that rotates the camera 180 or allows it to flip it when using Fluidd

**Describe alternatives you've considered**
A clear and concise description of how to achieve rotating a USB camera using MainsailOS.

kennethjiang, 2022-12-09:
This option is there already: https://obico.io/docs/user-guides/moonraker-obico/config/#webcam-section

Have you tried that?

OwendB1, 2022-12-09:
I could not find these settings when googling it for some reason. Turning on flip vertically and horizontally has resolved this for me. Thanks!

closed 2022-12-09 by OwendB1