TheSpaghettiDetective/obico-server #734: [Feature] Has anyone implemented an MQTT notification plug-in?

winnall, 2022-12-22:
**Is your feature request related to a problem? Please describe.**
I would like to link my 3D printer into my smarthome infrastructure, in particular to control a Hue light.

**Describe the solution you'd like**
Any events in TSD should publish themselves in an MQTT server defined by the user.

**Describe alternatives you've considered**
None. I already have and use MQTT, which is a commonly used and generic solution, and I’m not interested in further diversification.

**Additional context**
I consciously chose to use hosting by Obico rather than self-hosting. The plug-in that should be suitable for hosting there. I *could* program it myself, but I’d have to learn Python first.

winnall, 2022-12-22:
I’ve just realised that an MQTT notification plug-in should probably be in the same network as the MQTT server, at least in a smarthome context. Thus it is really only applicable to a self-hosted server.

For the Obico-hosted server, you’d probably want MQTT notifications to be generated by a client (in my case the Obico plug-in for OctoPrint).

kennethjiang, 2022-12-27:
For anyone who wants to take a stab at this task, we have a in-depth tutorial for building a notification plugin at: https://obico.io/docs/tutorials/pushover-notification-plugin/

jcgoette, 2024-05-15:
A few quick notes for future development:

- https://github.com/eclipse/paho.mqtt.python
- https://github.com/OctoPrint/OctoPrint-MQTT/blob/master/octoprint_mqtt/__init__.py

berkeoznalbant, 2025-01-10:
If you want to integrate obico to homeassistant I forked @johnc2k 's integration so that I get mobile notifications from homeassistant app about obico printer actions.

https://github.com/berkeoznalbant/Obico2HomeAssistant