TheSpaghettiDetective/obico-server #743: [Feature] Webhooks for print events

lnjustin, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
The existing notification options are fine for conventional approaches to being notified of a print failure on my phone or desktop, but integration with my smart home would expand the notifications options and enable smarter responses to print failures. For example, I can announce a failed print event over my Sonos speakers, change the color of smart bulbs for an unobtrusive notification, etc. Handling of print failure events could then further be dependent on time of day, presence, and the like. Especially if Obico gets into the safety space where the AI is able to detect safety-related events (e.g., fires), integration with smart homes would be invaluable, e.g., for triggering integrated sirens, triggering a security system event so a monitoring center is notified, cutting power to the printer via a smart switch, etc. The existing client API works to some extent, but requires constant polling of the Obico server. Very frequent polling would be required to obtain timely events. Webhooks would enable 'push' like notifications that are timely, without taxing the obico server with polling. 

**Describe the solution you'd like**
POSTing json content for print events via a webhook. This enhancement would allow the user to enter different webhook endpoints to be called for different print events (e.g.,  endpoint1 for print start, endpoint2 for print failure, endpoint3 for temp cooldown)?

**Describe alternatives you've considered**
Parsing email events is clunky and the reliability is questionable.

lnjustin, 2023-01-14:
I've updated the following files with a new 'webhook' notification plugin:
frontend/src/notifications/plugins.js
frontend/src/notifications/plugins/webhook.vue
backend/notifications/plugins/webhook/__init__.py

Am I missing anything? 

After updating those files on my hard drive, do I just need to run
docker compose up --build -d
as if upgrading the server (https://www.obico.io/docs/server-guides/ops/)

I tried that but the webhook plugin didn't show up on the frontend page, so I must be missing something....

kennethjiang, 2023-01-15:
@lnjustin Great that you are already working on it. Can you please open a PR against our "master" branch? We can continue our discussion there

kennethjiang, 2023-02-20:
Closed by https://github.com/TheSpaghettiDetective/obico-server/pull/758. will be released soon.

closed 2023-02-20 by kennethjiang