TheSpaghettiDetective/obico-server #745: ValueError: Unable to configure formatter 'console_debug' on tasks and web container

wieluk, 2023-01-18:
**Describe the bug**
On a fresh install with ```git clone -b release https://github.com/TheSpaghettiDetective/obico-server.git``` and ```cd obico-server && sudo docker compose up -d``` I get ```ValueError: Unable to configure formatter 'console_debug'``` in the tasks and web container.
I added two log files.

**To Reproduce**
Steps to reproduce the behavior:
1. fresh install

**Hosting environment (please complete the following information):**
 - OS: ```NAME="QTS"
VERSION="5.0.1 (20221215)"
ID=qts
PRETTY_NAME="QTS 5.0.1 (20221215)"
VERSION_ID="5.0.1"```
 - Hardware: Qnap Ts-364

**Additional context**
Add any other context about the problem here.
[_obico-server_tasks_1_logs.txt](https://github.com/TheSpaghettiDetective/obico-server/files/10443565/_obico-server_tasks_1_logs.txt)
[_obico-server_web_1_logs.txt](https://github.com/TheSpaghettiDetective/obico-server/files/10443566/_obico-server_web_1_logs.txt)

jodrell2000, 2023-01-19:
I'm seeing the same issue trying to run using an updated, then fresh install on a Mac Studio running MacOS 13.1

wieluk, 2023-01-19:
Do you know what is the last working version?

kennethjiang, 2023-01-21:
This is because the right command to upgrade a server is:

`sudo docker compose up --buld -d`

See [instructions](https://obico.io/docs/server-guides/ops/#upgrade-obico-server) for details.

Closing now. Feel free to re-open it if the this issue still exists after the correct upgrading command.

closed 2023-01-21 by kennethjiang