TheSpaghettiDetective/obico-server #747: Request to add container to unraid CA

hydazz, 2023-01-24:
Hi,
I am the maintainer of https://github.com/imagegenius/obico, which is a `obico-server` 'monolithic' container, specifically for unraid. I had sent a request to the Community applications admin to get the `templates` repo uploaded, which was accepted. but within the templates repo is also the template for obico. so currently my obico image is available from the CA. I hope that this is ok and I am happy to make any changes that are requested, to hopefully provide obico to uses of unraid without a janky docker-compose setup.

sorry to ask for forgiveness rather than permission here

Thanks

(don't snoop into the dockerfile, _some sacrifices had to be made..._)

kennethjiang, 2023-01-24:
Do you just need our okay for adding your Obico image to unraid CA? if so, you've got it. :) You actually didn't have to send this request. This repo is on AGPL license, and you have all the permissions that AGPL license will give you. :)

kennethjiang, 2023-01-24:
I'll close this issue for now. Feel free to reopen it if I misunderstood your request.

hydazz, 2023-01-30:
Thank you for clarifying this, that is what I thought but didn't want to hurt anyone's feelings, perhaps it would be beneficial to unraid users to add instructions to install obico-server to the `unraid_guide` on the docs? Unraid users won't use docker-compose unless it's absolutely necessary. but with my CA template, just search obico and it's a 1 click install,

kennethjiang, 2023-01-30:
Sure! Feel free to send in a PR for the doc update.

Please note that we need to have a way for the users to keep the server up to date. It's not clear how this will be done. But maybe after you submit a PR, it'll become more clear to me.

hydazz, 2023-01-30:
updating is done by updating the image, the obico-server files are stored in /app within the image and persistant data is symlinked to /config, like timelapses. so just pulling an updated image grabs the latest version. does that clear things up :). there are no commands to run or anything, simples

kennethjiang, 2023-02-02:
Will unraid server automatically pull the new image version? On Mac, Docker will not even check the image repo if an image has already been pulled and is available on the host. This is why most Docker image has "tag" attached to it, such as https://hub.docker.com/_/python/tags

hydazz, 2023-02-02:
unraid checks for new images and marks the container as having an update available, docker does not check whether an update is available, they have to be manually recreated, there are containers like watchtower that notify you when there's an update. 

> This is why most Docker image has "tag" attached to it

not entirely correct, in your example different tags are used to distinguish the specific version of python and the OS it runs on

overall running the image on unraid = notification to update (you can click update and it'll pull the latest image)
still simpler than running git pull + docker-compose

kennethjiang, 2023-02-07:
This is pretty cool. Wish the vanilla docker is this intelligent.

hydazz, 2023-02-07:
Yeah, docker desktop is mainly for development, you're probably not going to have a mac setup to run 24/7 for a server haha.
this is why most server OS's have their own update mechanisms for docker