TheSpaghettiDetective/obico-server #759: [Feature] Running Obico server as a Home Assistant add-on

iamkarlson, 2023-02-09:
**Is your feature request related to a problem? Please describe.**
Home Assistant is a powerful platform to integrate all sorts of hardware and software at once. Many people use it to run auxiliary software such as DNS servers, or grocery shopping lists. Since the format of the addon is very well-defined and based on docker images, it would be nice if you can distribute one. 

**Describe the solution you'd like**
Obico-supported HA addon

**Describe alternatives you've considered**
Creating my own hacked addon. 

**Additional context**
https://developers.home-assistant.io/docs/add-ons

MallocArray, 2023-02-09:
The obico-server isn't supported on Raspberry Pi hardware, as it needs more resources than a Pi can offer for doing the detection, and many HomeAssistant installs are done on a Pi.

The obico plugins for Octoprint or Moonraker/Klipper run on whatever device you are running those on typically and wouldn't be applicable to HA as far as I can tell.

nobodyguy, 2023-04-03:
So by looking more into it, I will go with the cleanest way - HA addon with REST ML API and HA integration implementing camera feed scanning service and providing camera entity with detected failures and maybe some  additional sensors with info about detected failures.
This way you can create your own HA automation like calling failure scanning service only when your printer is actually printing (info from PrusaLink or Octoprint integration) and sending standard HA notifications in case of detected failure.

ForceConstant, 2024-03-03:
Any updates on this? I would like this as an option.

kennethjiang, 2024-03-03:
We haven't been able to get around to it. Hope someone in the community can pitch in to implement it using our API, or even send a PR to our main repo.

ForceConstant, 2024-03-11:
@kennethjiang Ok assuming one has a self-hosted obico server already running. Can someone give me the high level primer on which API's would be used to send data to obico? As a start, I would assume you would create a dummy printer in obico web-app, and then HASS-plugin would send print/image data periodically, and also receive data. 

The API's listed at https://www.obico.io/docs/api/ seem to only be a way to receive data from obico and not send data.

ForceConstant, 2024-03-11:
@kennethjiang can you specifically tell me how to tell obico server there is a new image url to process?

ForceConstant, 2024-03-12:
@kennethjiang So looking more at this, it seems that this addon would be in the same mold as the current octoprint/klipper addons. Which seem to all talk to obico-server using undocumented API's. Such as '/api/v1/octo/pic/', etc. Or do I have that wrong?

kennethjiang, 2024-03-13:
> @kennethjiang So looking more at this, it seems that this addon would be in the same mold as the current octoprint/klipper addons. Which seem to all talk to obico-server using undocumented API's. Such as '/api/v1/octo/pic/', etc. Or do I have that wrong?

Ah yes we haven't got a chance to document the agent side's of the APIs. You will need to reverse-engine them but hopefuly they are not difficult to understand.

tannisroot, 2024-04-15:
> A lot of people run HA on x86 hardware like Intel NUC or older server/thin client PC.
> 
> However, by briefly looking into the dependencies I can see it wouldn't be as easy as wrapping the obico-server docker image in HA addon. Obico-server contains ml_api, web and redis services. These need to be either merged into a single Docker container or wrapped in an HA addon per every service (and there is no dependency management between HA addons). Both solutions are hacky and would require some effort. I can try the first solution, some work has been done here - https://github.com/fredrikbaberg/hassio-addons/blob/master/tsd/Dockerfile but I can't promise anything.

That sounds like something suitable for Kubernetes

nobodyguy, 2024-09-09:
> So by looking more into it, I will go with the cleanest way - HA addon with REST ML API and HA integration implementing camera feed scanning service and providing camera entity with detected [...]

Got some initial work done
[image]

ForceConstant, 2024-09-10:
> > So cool. Is your code available on github? If so, can you post the link here so that other community members can use it too?
> 
> HA addon would require a lot more work since HA addons are Docker images based on the specific base layer with S6 process supervisor and other additional stuff.

Can you explain what this means? There are plenty of integrations in the main HA branch, but also stuff in HACS https://hacs.xyz/ , and I don't think any of them require docker at all. What do you mean? I understand for Obico we would most likely need to run obico server separately from HA, but the integration itself would not need to be a container.

nobodyguy, 2024-09-12:
So PoC of the addon is here - https://github.com/nobodyguy/obico_ml_ha_addon
I'll work on the building and do some cleanup over the weekend.
HA integration should also be published during the weekend.