TheSpaghettiDetective/obico-server #760: Bulk Management of records

google007, 2023-02-09:
**Is your feature request related to a problem? Please describe.**
Cannot bulk delete/save/move records

**Describe the solution you'd like**
Menu options to allow the selecting a group of records similar to shift/ctrl

**Describe alternatives you've considered**
There are no alternatives, you can only delete records individually, after entering into the subfolder location where media file is.

**Additional context**
None.

kennethjiang, 2023-02-15:
Assuming by "records", you mean the print history. Bulk delete is now implemented in mobile app version 1.81 which will be released soon.

closed 2023-02-15 by kennethjiang

google007, 2023-02-15:
Thankyou Kenneth, I will take a look and give you feedback if any.

Jay
C: 604.700.9138
M: ***@***.******@***.***>