TheSpaghettiDetective/obico-server #761: Obico Self Hosted Server on NAS

jjensen6823, 2023-02-11:
I have managed to install GIT, Docker and Docker-composer on my Terra Master NAS, and I cloned the repository and used composer to build the containers.  Docker says there are four containers running: obico-server_tasks_1, obico-server_web_1, obico-server_ml_api_1, and obico-server_redist_1,  Since it's an NAS, connecting to the server via localhost isn't really an option, and when I attempt to connect to 192.168.8.100:3334 or 192.168.8.100:3334/admin, I get "Site cannot be reached.  Server took too long to respond."  I installed the web server app on my NAS, but am unsure how I would configure it.  I tried creating a virtual server on port 3334, and it said it was already occupied.  I don't know if something on the NAS is blocking the connection or if there is something on the server I need to configure.  I realize NAS config is not an Obico issue, but if anyone could point me in the right direction to look, I'd be grateful.

MallocArray, 2023-02-11:
That is odd, because if the containers are up and you didn't make any changes to the files in the repository, then it should be up and running at the IP address of your NAS:3334  as you tried.  I haven't ran it on a NAS, so not sure if they have a firewall or anything that would need that port to be opened.

As an alternative you could consider trying to run it on a free Oracle Cloud Infrastructure instance. I have a guide for it here:
https://github.com/MallocArray/obico-server

jjensen6823, 2023-02-11:
I may resort to that, but I'd rather run it locally.  I don't understand why it's not running, either.

jdoncouse, 2023-02-12:
Confirming it's not working on my arch server w/docker compose
Logs for obico-server-tasks) 

return self.handle_argv(self.prog_name, argv[1:])
  File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 491, in handle_argv
    return self.execute(command, argv)
  File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 419, in execute
    ).run_from_argv(self.prog_name, argv[1:], command=argv[0])
  File "/usr/local/lib/python3.7/site-packages/celery/bin/worker.py", line 223, in run_from_argv
    return self(*args, **options)
  File "/usr/local/lib/python3.7/site-packages/celery/bin/base.py", line 253, in __call__
    ret = self.run(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/celery/bin/worker.py", line 258, in run
    **kwargs)
  File "/usr/local/lib/python3.7/site-packages/celery/worker/worker.py", line 99, in __init__
    self.app.loader.init_worker()
  File "/usr/local/lib/python3.7/site-packages/celery/loaders/base.py", line 114, in init_worker
    self.import_default_modules()
  File "/usr/local/lib/python3.7/site-packages/celery/loaders/base.py", line 108, in import_default_modules
    raise response
  File "/usr/local/lib/python3.7/site-packages/celery/utils/dispatch/signal.py", line 288, in send
    response = receiver(signal=self, sender=sender, **named)
  File "/usr/local/lib/python3.7/site-packages/celery/fixups/django.py", line 84, in on_import_modules
    self.worker_fixup.validate_models()
  File "/usr/local/lib/python3.7/site-packages/celery/fixups/django.py", line 122, in validate_models
    self.django_setup()
  File "/usr/local/lib/python3.7/site-packages/celery/fixups/django.py", line 118, in django_setup
    django.setup()
  File "/usr/local/lib/python3.7/site-packages/django/__init__.py", line 19, in setup
    configure_logging(settings.LOGGING_CONFIG, settings.LOGGING)
  File "/usr/local/lib/python3.7/site-packages/django/utils/log.py", line 76, in configure_logging
    logging_config_func(logging_settings)
  File "/usr/local/lib/python3.7/logging/config.py", line 800, in dictConfig
    dictConfigClass(config).configure()
  File "/usr/local/lib/python3.7/logging/config.py", line 546, in configure
    'formatter %r' % name) from e
ValueError: Unable to configure formatter 'console_debug'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/logging/config.py", line 384, in resolve
    found = self.importer(used)
ModuleNotFoundError: No module named 'colorlog'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/logging/config.py", line 543, in configure
    formatters[name])
  File "/usr/local/lib/python3.7/logging/config.py", line 654, in configure_formatter
    result = self.configure_custom(config)
  File "/usr/local/lib/python3.7/logging/config.py", line 469, in configure_custom
    c = self.resolve(c)
  File "/usr/local/lib/python3.7/logging/config.py", line 397, in resolve
    raise v
  File "/usr/local/lib/python3.7/logging/config.py", line 384, in resolve
    found = self.importer(used)
ValueError: Cannot resolve 'colorlog.ColoredFormatter': No module named 'colorlog'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/bin/celery", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/site-packages/celery/__main__.py", line 16, in main
    _main()
  File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 322, in main
    cmd.execute_from_commandline(argv)
  File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 499, in execute_from_commandline
    super(CeleryCommand, self).execute_from_commandline(argv)))
  File "/usr/local/lib/python3.7/site-packages/celery/bin/base.py", line 305, in execute_from_commandline
    return self.handle_argv(self.prog_name, argv[1:])
  File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 491, in handle_argv
    return self.execute(command, argv)
  File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 419, in execute
    ).run_from_argv(self.prog_name, argv[1:], command=argv[0])
[... 4 lines cut ...]
  File "/usr/local/lib/python3.7/site-packages/celery/bin/worker.py", line 258, in run
    **kwargs)
  File "/usr/local/lib/python3.7/site-packages/celery/worker/worker.py", line 99, in __init__
    self.app.loader.init_worker()
  File "/usr/local/lib/python3.7/site-packages/celery/loaders/base.py", line 114, in init_worker
    self.import_default_modules()
  File "/usr/local/lib/python3.7/site-packages/celery/loaders/base.py", line 108, in import_default_modules
    raise response
  File "/usr/local/lib/python3.7/site-packages/celery/utils/dispatch/signal.py", line 288, in send
    response = receiver(signal=self, sender=sender, **named)
  File "/usr/local/lib/python3.7/site-packages/celery/fixups/django.py", line 84, in on_import_modules
    self.worker_fixup.validate_models()
  File "/usr/local/lib/python3.7/site-packages/celery/fixups/django.py", line 122, in validate_models
    self.django_setup()
  File "/usr/local/lib/python3.7/site-packages/celery/fixups/django.py", line 118, in django_setup
    django.setup()
  File "/usr/local/lib/python3.7/site-packages/django/__init__.py", line 19, in setup
    configure_logging(settings.LOGGING_CONFIG, settings.LOGGING)
  File "/usr/local/lib/python3.7/site-packages/django/utils/log.py", line 76, in configure_logging
    logging_config_func(logging_settings)
  File "/usr/local/lib/python3.7/logging/config.py", line 800, in dictConfig
    dictConfigClass(config).configure()
  File "/usr/local/lib/python3.7/logging/config.py", line 546, in configure
    'formatter %r' % name) from e
ValueError: Unable to configure formatter 'console_debug'

kennethjiang, 2023-02-15:
> Confirming it's not working on my arch server w/docker compose. I've ran this in the past without issues on the same server. Logs for obico-server-tasks and obico-server-web-1_logs
> 
> [_obico-server-tasks-1_logs.txt](https://github.com/TheSpaghettiDetective/obico-server/files/10716857/_obico-server-tasks-1_logs.txt) [_obico-server-web-1_logs.txt](https://github.com/TheSpaghettiDetective/obico-server/files/10716858/_obico-server-web-1_logs.txt)

Based on your log, you probably used the wrong command to upgrade the server. For the correct command, check: https://obico.io/docs/server-guides/ops/#upgrade-obico-server

kennethjiang, 2023-02-15:
> I may resort to that, but I'd rather run it locally. I don't understand why it's not running, either.

This sounds like an environment issue. I'm closing this for now unless you have a reason to believe this is a bug. Please go to https://obico.io/discord and you will be able to get help more quickly.

closed 2023-02-15 by kennethjiang