TheSpaghettiDetective/obico-server #764: [Feature] Run Obico without Octoprint or Klipper

pniewiadowski, 2023-02-22:
**Is your feature request related to a problem? Please describe.**
I don't have Octoprint or Klipper running but still want to monitor prints and get notification from Obico if something went wrong.

**Describe the solution you'd like**
I'd like to run Obico without Octoprint or Klipper. Preferably with Camera RTSP stream

kennethjiang, 2023-02-23:
Have you tried the workaround of running octoprint with a virtual printer? If so, please let us know where this workaround is still falling short of so that we have better idea on what needs to be done here.

pniewiadowski, 2023-02-23:
I was trying that but while I have RTSP camera I was failing to set it up correctly in Octoprint, therefore I wasn't able to see the camera stream in Obico.

kennethjiang, 2023-02-27:
There is a chance that RTSP camera can work with our streaming, at least in theory. However, this situation seem rare among our users so we may not be able to allocate resources to it any time soon. In the mean while, feel free to take a look at our source code to see if you can figure this out yourself so that you don't have to wait for a long time for us to get around to it, if at all.

kevinavos, 2024-01-07:
This feature would be useful for Bambu P1S users. You can stream a P1S camera using: https://github.com/slynn1324/BambuP1Streamer?tab=readme-ov-file so if we were able to use this to stream directly to Obico to get a notification if something went wrong that would be lovely.

nvtkaszpir, 2024-01-29:
I believe people want to run obico for image detection without any other tools.

Running with different camera types (video streams instead of image capture) is a request for another feature.

skl111, 2024-09-26:
I would love this for all of the Bambu Printers as I currently use OctoEverywhere for my Bambu fleet due it being able to run standalone.

kennethjiang, 2024-09-26:
> I would love this for all of the Bambu Printers as I currently use OctoEverywhere for my Bambu fleet due it being able to run standalone. I have 1600 unused Obico AI hours due to my support of this project.

Thank you for your support of our project! I know quite many Obico users have had luck with using https://github.com/jneilliii/OctoPrint-BambuPrinter to integrate Obico on their Bambu printers. The reason why we haven't tried to come up with a standalone solution for Bambu printers is that I don't want to duplicate the awesome job jneilliii has done. Of course if for some reason that plugin doesn't work for you, definitely let me know and I will consider developing our own.

skl111, 2025-01-07:
> Thank you for your support of our project! I know quite many Obico users have had luck with using https://github.com/jneilliii/OctoPrint-BambuPrinter to integrate Obico on their Bambu printers. The [...]

So it doesn't work because I don't want to run multiple PI's nor do I want to run Multiple Octoprint Instances so I can view the camera. I would need to use OctoDeploy(which is great itself) to do that again. For me less is more. Thats just my preference. I still have my Enders which I am retrofitting with linear rails and the Mercury kits and voron dragonburner setups, those I will be using octodeploy with octoprint and obico once I get them back online.

nvtkaszpir, 2025-01-07:
Some time ago I extracted obico ml API as a standalone app without any authorization or whatever. Maybe someone will find it more convenient to use.

https://github.com/nvtkaszpir/3d-print/tree/main/obico-ml-api-only

saladpanda, 2025-01-08:
@nvtkaszpir That only works with the self-hosted version of obico, correct?

Like @skl111 i want to feed images or a camera stream (from my bambulab printer) to the paid Obico Cloud where I still have almost 1k detection hours.

nvtkaszpir, 2025-01-08:
@saladpanda yes this is only for self hosted.
It has nothing to do with Obico Cloud.

Deskwared, 2025-03-16:
> Have you tried the workaround of running octoprint with a virtual printer? If so, please let us know where this workaround is still falling short of so that we have better idea on what needs to be done here.

I have a Snapmaker J1 that I’d like to have monitored by Obico.  Of course, I cannot initiate prints from octoprint, since the touchscreen runs some important commands. I’ve seen references to running octoprint with a virtual printer, to make Obico work (here and elsewhere) but have not actually found any instructions on how to accomplish this.  I imagine it is a case of initiating the “print” on the virtual printer at the same time that the locally initiated print starts running, but I’m not sure how to do that.  Any guidance would be appreciated!

skl111, 2025-06-23:
With the recent firmware lockdown that Bambulab implemented, you have a chance to get uses using the LAN/DEV mode for cameras and you could send prints via the LAN/DEV Mode as well utilizing AMS mapping, enabling features for those of us who don't mind running a local relay to the Obico cloud or a Local Obico instance.

I have a printer or 2 I can allocate for testing.

kennethjiang, 2025-06-23:
Unfortunately, the effort to build such a system to work around the proprietary systems is too high. And the risk of being broken any time is too great to justify the effort. But if someone can build it out, we will be happy to merge a PR.

skl111, 2025-07-12:
> > With the recent firmware lockdown that Bambulab implemented, you have a chance to get quite a few users using the LAN/DEV mode for cameras and you could send prints via the LAN/DEV Mode as well utilizing AMS mapping, enabling features for those of us who don't mind running a local relay to the Obico cloud or a Local Obico instance.
> > I have a printer or 2 I can allocate for testing.
> 
> Unfortunately, the effort to build such a system to work around the proprietary systems is too high. And the risk of being broken any time is too great to justify the effort. But if someone can build it out, we will be happy to merge a PR.

Thats strange considering that there are projects for HA that can use lan mode and Orca uses lan mode, but of course I don't know what the Obico's dev team's time is worth as I don't know the metrics.

That as they say is that.