TheSpaghettiDetective/obico-server #766: [BUG] Mangled video with self-hosted server & Crowsnest

whi-tw, 2023-02-27:
**Describe the bug**
Video stream appears mangled in obico-server. Unsure if this is a me-problem, but the stream works fine when viewed directly in the browser (http://klipper/webcam?action=stream) but the stream in the obico-server view is garbage after loading (the initial snapshot looks fine). Looks to possibly be an encoding problem, maybe?

**To Reproduce**
Steps to reproduce the behavior:
1. Open the app, observe that the webcam snapshot looks ok
2. Wait for the stream to start
3. Observe that the stream looks like garbage

**Screenshots**
[image]

**Hosting environment (please complete the following information):**
 - OS: debian
 - Version:
   - [moonraker-obico#2e2cefa](https://github.com/TheSpaghettiDetective/moonraker-obico/tree/2e2cefa6f9d8cb565e7040ab82641514b4566643)
   - [obico-server#72b26df](https://github.com/TheSpaghettiDetective/obico-server/tree/72b26dfa6f35800388816c621870664e5501dd19)
   - [mainsail-crew/crowsnest#5418550](https://github.com/mainsail-crew/crowsnest/tree/541855054f6de4dfa8ff4d5e73dd36b213049903)
 - Hardware: PC

**Additional context**
I've tested this, and the stream is broken on MacOS (Safari / Chromium) and in the iOS app.

crowsnest camera config:

```
[cam top]
mode: mjpg                              # mjpg/rtsp
port: 8080                              # Port
device: /dev/video0
resolution: 1440x1080                   # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
```

MallocArray, 2023-02-27:
This has been observed with high resolutions.  Try something lower than 800x600 and I believe it will work.  I'm using 640x480 and it is operational.  I believe Kenneth mentioned a fix is in process

joshua76y, 2023-10-14:
I have the same problem，the webcam window is black and change resolution to 640*480 can't fit

kennethjiang, 2023-10-14:
@joshua76y Are you using a Raspberr Pi? Also it's a lot easier if you come to our discord channel since more people will see your message and help you: https://obico.io/discord-obico-klipper/