TheSpaghettiDetective/obico-server #774: [Feature] Sync AI Detection to Octolapse for cleaner frames

OstlerDev, 2023-03-27:
### **Is your feature request related to a problem? Please describe.**
My webcam sits above my printhead meaning the print bed and item printing is often obscured from view. This makes the AI detection not find obvious spaghetti cases because it will take photos at random.

### **Describe the solution you'd like**
If Obico was able to use the pictures taken by Octolapse as input to the AI Algorithm. Octolapse automatically moves the print head out of the way before it takes a picture, if Obico could use those pictures it would have access to an un-obscured view of the print in progress.

### **Describe alternatives you've considered**
- Moving the print head off to the side using gcode for multiple extra seconds to increase odds of spaghetti being caught.
- Increase the "deshake" delay in Octolapse from 150ms to 3000ms to allow Obico to get a clear view of the print for at least three full seconds each layer.

### **Additional context**
Because my printer is fully enclosed there are no other positions that would give Obico a good view of the print, the only space in my printer for a camera was above the print-head.

I have included two timelapses for context, one taken by Obico, and the second taken by Octolapse. In the Obico timelapse you can witness that the AI did properly detect that there was spaghetti on the bed a couple times, however because the print head obscured the print after that the confidence would always go down. You can see how the Octolapse frames are a much clearer source for the AI to process.

### Obico Frames Example
https://user-images.githubusercontent.com/1490586/227826313-5eec9d9f-07cd-4309-a198-d6195dbfbe68.mp4

### Octolapse Frames Example
https://user-images.githubusercontent.com/1490586/227826955-47fda07a-762f-4ae5-83cd-42ff5b87e080.mp4

kennethjiang, 2023-03-31:
We are cooking something related to this. Stay tuned! :)

OstlerDev, 2023-04-01:
Perfect! Looking forward to what might be in store :)