TheSpaghettiDetective/obico-server #782: [Feature] Dynamic Values for Webhooks (Push Notifications)

Xosten, 2023-04-18:
**Is your feature request related to a problem? Please describe.**

As a user relying on NFTY for sending notifications, if I want to use it with my Obico instance, I can only send a defined message that isn't dynamic.

**Describe the solution you'd like**

As a user, when I set up my webhook I would like to be able to add dynamic values so I could have personalised notifications sent based on the related matter.

For example: `https://ntfy.sh/topic/send?message={$OBICO-MESSAGE}&title={$OBICO-TITLE}&priority=3&tags=bell`

Whereas for this example if the `Benchy` print is ready : 
- $OBICO-TITLE = Print Ready
- $OBICO-MESSAGE = Your Benchy is ready

Or the print `Benchy` started:
- $OBICO-TITLE = Print Started
- $OBICO-MESSAGE = Benchy print in progress ETA 1:20PM

We could see more variables available to the user in order to customise as much as possible, like: 
- Printer Name
- AI Watch Status
- File Name

**Describe alternatives you've considered**
None at the current stage

**Additional context**
NTFY is not the only notification system we can see for this context, Gotify could also be another option. 

iOS users that are self-hosting don't have access to print progress notifications. Webhooks could be the solution to bring such alerts to those users.

We could see the same as the Discord notification, where the difference lies in the ability to structure in a certain manner the messages.

kennethjiang, 2023-04-18:
I like the idea behind it. But your proposal seems to be too limiting and specific to NTFY.

I'd like to propose these variable substitutes for now:

- `EVENT`: "print-started", "print-finished", "failure-alert", etc.
- `PRINTER_NAME`
- `GCODE_NAME`

Does it work for you @Xosten

Xosten, 2023-04-18:
Thanks Kenneth for your quick reply. 

Yes totally works for me, I mentioned NTFY as I’m using it personally. If it works with any Webhook, that’s would be awesome! We can then see later if other people see other variables or use cases 😉. 

Cheers !

LeSauvage, 2023-08-05:
Is this feature under development? I've made a basic plugins working with NTFY, I could pursue it more if you guys require specific features / variables