TheSpaghettiDetective/obico-server #785: [BUG] moonraker_obico.app process eating my Pi Memory as the print progress!

izGHLooL, 2023-04-23:
recently my prints keep getting stopped out of the blue , I tried multiple files and slicers but with no luck, I started looking around and found out that my Pi memory is being full by one process which is "moonraker_obico.app" pictures attached

**Screenshots**
[image]
[image]
[image]

**Hosting environment (please complete the following information):**
 - OS: Klipper
 - Version latest
 - Hardware Voron 2.4 r2

kennethjiang, 2023-04-24:
This looks like a memory leak. I'd like to figure out if the memory leak happens only when the printer is printing. Can you let your printer sit idle for a while to see if the memory leak will still happen?

izGHLooL, 2023-04-24:
hi @kennethjiang ,
I hope you are doing well.
even when the printer is set idle with nothing printing it will do the same.

I recently installed the Celestrius Data Collection add-on since I already have a nozzle cam installed but unfortunately, I could not register.
so what I did was uninstalled Mooraker-obico app as well as Celestrius and my memory returned to normal behavior.

now I don't have obico remote monitoring installed, meaning I can not monitor prints remotely. until I figure out what is the problem.

kennethjiang, 2023-04-30:
Sorry for whatever reason all github emails went to my spam folder.

I remember an old Obico version used to have a memory leak. Can you check if your moonraker-obico is the latest version?

izGHLooL, 2023-04-30:
Yes I’m on the latest version of everything

kennethjiang, 2023-04-30:
Can you send an email to support@obico.io so that we can set up a troubleshoot video call?

izGHLooL, 2023-04-30:
> Can you send an email to ***@***.*** so that we can set up a
> troubleshoot video call?

Ok sure , let me reach my work , and i will send the email