TheSpaghettiDetective/obico-server #786: [BUG?] Mails from Obico sent to Apple are not showing camera snapshot directly

nicktodorov, 2023-05-01:
**Describe the bug**
For some reason, notification mails from Obico self hosted server do not show the camera snapshot. 
There is message "Your network preferences prevent content from loading privately.". But the problem is, that in order to make this work as supposed, I need to turn off completely Private relay feature, which I do not want for obvious reasons. 
Since this is the only email that has this issue (even span loads correctly), there should be something specific on how Obico emails are working.
Apologies if this is a setting I have missed and not a bug, but I could not find anything on this topic specific to Obico.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup Obico self hosted server
2. Configure email notifications
3. Open notification email on an Apple device. It doesn't matter what mail client or device is used, if there is Private Relay (hiding your IP address for email tracking), the problem is reproducible.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
 - OS: iOS, Mac OS
 - Version: 16.4.1, Ventura 13.3.1
 - Hardware: Mac, iPhone, iPad

nicktodorov, 2023-05-02:
Where do I have to set this SSL? In docker, in Django or the ngrok?

kennethjiang, 2023-05-06:
@nicktodorov Please refer to other sys admin resources for how you can set up SSL.

I'm closing this issue as managing SSL is outside the scope of the Obico server. If there is reason to believe this is really a bug, please come back and reopen it.

closed 2023-05-06 by kennethjiang