TheSpaghettiDetective/obico-server #798: Error installing pip when doing "docker compose up - d" on Ubuntu 22.10.

Istria1704, 2023-05-26:
Hi, 

I'm trying to install Obico on Ubuntu Server 22.10 using the official guide. But during the final step, "docker compose up - d", I get an error and the process stops there:
=> ERROR [tasks 4/5] RUN pip install -U pip     1.7s 

Here is the console log. Thanks in advance for the help! 

`root@Obico:~/obico-server# docker compose up -d
[+] Building 29.8s (13/19)                            
 => [ml_api internal] load build definition fro  0.1s
 => => transferring dockerfile: 270B             0.0s
 => [ml_api internal] load .dockerignore         0.0s
 => => transferring context: 2B                  0.0s
 => [tasks internal] load build definition from  0.1s
 => => transferring dockerfile: 177B             0.0s
 => [tasks internal] load .dockerignore          0.1s
 => => transferring context: 118B                0.0s
 => [ml_api internal] load metadata for docker.  0.8s
 => [tasks internal] load metadata for docker.i  0.8s
 => CANCELED [ml_api 1/6] FROM docker.io/thespa  2.1s
 => => resolve docker.io/thespaghettidetective/  0.1s
 => => sha256:d43c180de04e03e8d0b69 743B / 743B  0.0s
 => => sha256:6eaa0374427cf04ad 9.70kB / 9.70kB  0.0s
 => => sha256:34667c7e4631207 43.56MB / 43.56MB  1.7s
 => => sha256:61b24721bca7180e5 3.04kB / 3.04kB  0.0s
 => => sha256:d18d76a881a47e51f4210 852B / 852B  0.3s
 => => sha256:119c7358fbfc2897ed635 529B / 529B  0.4s
 => => sha256:2aaf13f3eff07aa25f738 169B / 169B  0.5s
 => => sha256:4d96b2dafaa531e30 6.98MB / 6.98MB  1.6s
 => => sha256:f8c41b380cab6 406.87kB / 406.87kB  1.6s
 => => sha256:d2c1b485844605305214f 184B / 184B  1.8s
 => => extracting sha256:34667c7e4631207d64c99  26.9s
 => => sha256:60930c6a628c1ab0f118a00 0B / 92B  28.7s
 => => sha256:ba985066a7db79c080 0B / 523.69MB  28.7s
 => [ml_api internal] load build context         0.0s
 => => transferring context: 1.02kB              0.0s
 => [tasks 1/5] FROM docker.io/thespaghettidete  0.0s
 => [tasks internal] load build context          0.1s
 => => transferring context: 15.03kB             0.0s
 => CACHED [tasks 2/5] WORKDIR /app              0.0s
 => CACHED [tasks 3/5] ADD . /app                0.0s
 => ERROR [tasks 4/5] RUN pip install -U pip     1.7s
`

kennethjiang, 2023-05-29:
This looks like a networking/environment issue. Please seek help in our discord server: https://obico.io/discord/. Or you can try to run the command again

closed 2023-05-29 by kennethjiang

Istria1704, 2023-05-29:
Solution was to remove apparmor and reboot.

'''apt purge apparmor apparmor-utilities
reboot''' 

Then it worked fine.

I noticed the docker hello world also didn't work before that. So it was purely a docker problem and nothing to do with Obico.
I guess it has something to do with Docker on Ubuntu in an LXC container. But in Debian, Docker worked fine. Without removing apparmor. So not sure what the difference is between Debian and Ubuntu.

kennethjiang, 2023-05-29:
Gotcha. Thank you for the update!

keteflips, 2023-09-25:
Hello, same problem here, LXC in Proxmox too.

` => ERROR [ml_api 4/9] RUN pip install --upgrade pip                                                                                 13.1s
------
 > [ml_api 4/9] RUN pip install --upgrade pip:
1.360 Requirement already satisfied: pip in /usr/local/lib/python3.8/dist-packages (23.1.2)
1.649 Collecting pip
1.705   Downloading pip-23.2.1-py3-none-any.whl (2.1 MB)
1.866      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 13.5 MB/s eta 0:00:00
2.475 Installing collected packages: pip
2.476   Attempting uninstall: pip
2.477     Found existing installation: pip 23.1.2
2.600     Uninstalling pip-23.1.2:
8.137 ERROR: Could not install packages due to an OSError: [Errno 22] Invalid argument: '/usr/local/lib/python3.8/dist-packages/pip-23.1.2.dist-info/'
8.137
------
failed to solve: process "/bin/sh -c pip install --upgrade pip" did not complete successfully: exit code: 1
root@obico:~/obico-server# cat /etc/issue
Ubuntu 22.04.3 LTS \n \l
`

Istria1704, 2023-09-25:
Have you tried removing apparmor like written in the last message before yours?

keteflips, 2023-09-25:
> Have you tried removing apparmor like written in the last message before yours?

Yes I did.

Istria1704, 2023-09-25:
Hmm, that was wat solved it for me. Did you reboot the container afterwards?
And does the docker hello world also not work?

keteflips, 2023-09-25:
> Hmm, that was wat solved it for me. Did you reboot the container afterwards? And does the docker hello world also not work?

Yes, I restarted and the Hello World works well.

keteflips, 2023-09-25:
> Then I'm afraid I don't know a solution. Maybe you could try Debian instead of Ubuntu?

Exactly the same error in Debian :(

`
root@obico2:~/obico-server# sudo docker compose up -d
[+] Building 20.3s (8/13)                                                                                                 docker:default
 => [ml_api internal] load build definition from Dockerfile                                                                         0.1s
 => => transferring dockerfile: 524B                                                                                                0.0s
 => [ml_api internal] load .dockerignore                                                                                            0.2s
 => => transferring context: 70B                                                                                                    0.0s
 => [ml_api internal] load metadata for docker.io/thespaghettidetective/ml_api_base:1.3                                             1.3s
 => [ml_api 1/9] FROM docker.io/thespaghettidetective/ml_api_base:1.3@sha256:6d56828703051ae8116eea16a5ef11fe7ea25c3a834e13d65d498  0.0s
 => [ml_api internal] load build context                                                                                            0.2s
 => => transferring context: 879B                                                                                                   0.0s
 => CACHED [ml_api 2/9] WORKDIR /app                                                                                                0.0s
 => CACHED [ml_api 3/9] ADD . /app                                                                                                  0.0s
 => ERROR [ml_api 4/9] RUN pip install --upgrade pip                                                                               17.9s
------                                                                                                                                   
 > [ml_api 4/9] RUN pip install --upgrade pip:                                                                                           
1.607 Requirement already satisfied: pip in /usr/local/lib/python3.8/dist-packages (23.1.2)                                              
1.874 Collecting pip                                                                                                                     
1.947   Downloading pip-23.2.1-py3-none-any.whl (2.1 MB)                                                                                 
2.082      ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 16.1 MB/s eta 0:00:00                                                     
2.596 Installing collected packages: pip
2.596   Attempting uninstall: pip
2.597     Found existing installation: pip 23.1.2
2.720     Uninstalling pip-23.1.2:
12.63 ERROR: Could not install packages due to an OSError: [Errno 22] Invalid argument: '/usr/local/lib/python3.8/dist-packages/pip-23.1.2.dist-info/'
12.63 
------
failed to solve: process "/bin/sh -c pip install --upgrade pip" did not complete successfully: exit code: 1
root@obico2:~/obico-server# 

`

Istria1704, 2023-09-25:
Just guessing now. But maybe without "sudo"?
I install and run everything as the root user and never use sudo. It even gives an error if I do.
I guess you made a second user for docker?

Disclaimer: I'm relatively n00b with linux. So this method might not be the most secure from what I (barely) understand.

keteflips, 2023-09-26:
Can you tell me which linux distro and versión did you have?

Istria1704, 2023-09-26:
```Ubuntu 22.10 Obico lxc/tty1

Obico login: root
Password: 
Welcome to Ubuntu 22.10 (GNU/Linux 5.15.108-1-pve x86_
64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
Last login: Thu Jun 22 13:58:15 UTC 2023 on lxc/tty1
root@Obico:~# cat /etc/os-release
PRETTY_NAME="Ubuntu 22.10"
NAME="Ubuntu"
VERSION_ID="22.10"
VERSION="22.10 (Kinetic Kudu)"
VERSION_CODENAME=kinetic
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms
-and-policies/privacy-policy"
UBUNTU_CODENAME=kinetic
LOGO=ubuntu-logo
root@Obico:~# ```

Istria1704, 2023-09-26:
By the way: Did you enable "nesting" in the LXC container settings? I believe this is needed for docker to work.

Options => Features => Nesting

puterboy, 2023-09-27:
I am having the exact same problem on Ubuntu 18.04 when I tried a fresh install of obico-server.
The same sequence of:
```git clone -b release https://github.com/TheSpaghettiDetective/obico-server.git
cd obico-server
docker compose up -d```
worked for me last year but fails now with the error message:
``` => ERROR [tasks 4/5] RUN pip install -U pip```

puterboy, 2023-09-27:
I was able to get it to work by cleaning out all my docker files and starting again. Must have been something messed up with my docker setup.

Note that I needed to do "sudo docker compose up -d" the first time to get it to build

Istria1704, 2023-09-27:
Now that you mentioned it, I think I also had to a 'docker system prune' after removing apparmor and apparmor-utilities and rebuild the obico image.

keteflips, 2023-09-28:
I finally simply commented  pip install --upgrade pip in all the Dockerfile, because I already have the last version of pip, its not necessary to upgrade it.