TheSpaghettiDetective/obico-server #810: [BUG] Camera set to be rotated 180 in klipper.  Timelapse is upside down.

xanderphillips, 2023-08-01:
**Describe the bug**
Timelapse recorded from klipper enabled webcam that is configured to invert and flip video due to being mounted upside down is not recorded to timelapse right-way-around.

**To Reproduce**
Steps to reproduce the behavior:
With printer camera mounted upside down on gantry and Klipper configured for webcam using Adaptive MJPEG-Streamer (experimental), and set to have: Flip webcam image horizontally and vertically both checked.  This renders the video in mainsail correctly right side up and left to right, but not in obico-server / timelapse recordings.  I am able to rotate displayed video in the obico-server website, but timelapse is still recorded upside down.

**Hosting environment (please complete the following information):**
 - OS: debian x64

**Additional context**
Add any other context about the problem here.

neilhailey, 2023-08-01:
@xanderphillips You can override the webcam settings to flip the webcam in Obico mobile/web apps, and they will not impact Klipper/mainsail. 

The displayed video rotation in-app will not impact the timelapse, but flipping the webcam in the moonraker-obico.cfg will. Here is an example configuration: https://www.obico.io/docs/user-guides/moonraker-obico/config/.

xanderphillips, 2023-08-15:
> @xanderphillips You can override the webcam settings to flip the webcam in Obico mobile/web apps, and they will not impact Klipper/mainsail.
> 
> The displayed video rotation in-app will not impact the timelapse, but flipping the webcam in the moonraker-obico.cfg will. Here is an example configuration: https://www.obico.io/docs/user-guides/moonraker-obico/config/. Let me know if that doesn't work for you!

That works, thanks.

closed 2023-08-15 by xanderphillips