TheSpaghettiDetective/obico-server #818: [BUG] Brief bug description here

burak1000, 2023-09-18:
Janus gives error i tried many thing 

`

2023-09-19 03:23:52,075      INFO  obico.app - detected state change: Offline -> Operational
2023-09-19 03:23:52,077     DEBUG  obico.janus - Using camera streamer RSTP? False
2023-09-19 03:23:52,079     DEBUG  obico.server_conn - Sending to server: 
{'current_print_ts': -1, 'status': {'_ts': 1695065032.077938, 'state': {'text': 'Operational', 'flags': {'operational': True, 'paused': False, 'printing': False, 'cancelling': False, 'pausing': False, 'error': False, 'ready': True, 'closedOrError': False}, 'error': None}, 'currentZ': -0.021999789473684223, 'job': {'file': {'name': None, 'path': '', 'display': None, 'obico_g_code_file_id': None}, 'estimatedPrintTime': None, 'user': None}, 'progress': {'completion': 0.0, 'filepos': 0, 'printTime': 0.0, 'printTimeLeft': None, 'filamentUsed': 0.0}, 'temperatures': {'extruder': {'actual': 26.04, 'offset': 0, 'target': 0.0}, 'heater_bed': {'actual': 0.0, 'offset': 0, 'target': None}}, 'file_metadata': {'analysis': {'printingArea': {'maxZ': None}}, 'obico': {'totalLayerCount': None}}, 'currentLayerHeight': None, 'currentFeedRate': 1.0, 'currentFlowRate': 1.0, 'currentFanSpeed': 0.0}}
2023-09-19 03:23:52,080      INFO  obico.janus - Starting webcam streamer
2023-09-19 03:23:52,081     DEBUG  obico.moonraker_conn - POST http://127.0.0.1:7125/server/database/item
2023-09-19 03:23:52,087     DEBUG  obico.janus - Popen: /root/moonraker-obico/moonraker_obico/bin/janus/setup.sh -A 652accbfdab01fd84b1d -V true
2023-09-19 03:23:52,096     DEBUG  urllib3.connectionpool - Starting new HTTP connection (1): 127.0.0.1:7125
2023-09-19 03:23:52,099   WARNING  obico.webcam_stream - Not running on a Pi. Quitting video_pipeline.
2023-09-19 03:23:52,119     DEBUG  urllib3.connectionpool - http://127.0.0.1:7125 "POST /server/database/item HTTP/1.1" 200 74
2023-09-19 03:23:52,230   WARNING  obico.janus - Janus setup failed. Skipping Janus connection. Error: 
b"dpkg-query: package 'janus' is not installed\nUse dpkg --contents (= dpkg-deb --contents) to list archive files contents.\n"
2023-09-19 03:23:53,097      INFO  backoff - Backing off wait_for_janus(...) for 0.3s (ConnectionRefusedError: [Errno 111] Connection refused)
2023-09-19 03:23:54,352      INFO  backoff - Backing off wait_for_janus(...) for 0.5s (ConnectionRefusedError: [Errno 111] Connection refused)
`

burak1000, 2023-09-18:
it looks like give error when run setup.sh. i installed janus via snap too . but it not helped

wtih 
Klipper - Moonraker

kennethjiang, 2023-09-19:
This looks like an environment issue. Please come to our obico for klipper community channel to get speedy help: https://www.obico.io/discord-obico-klipper/

burak1000, 2023-09-19:
firstly thank you so much for your helping. 
i solve the problem 
[image]

first i updated everything from here.
and i add webcam section in my moonraker.conf file 

`

[webcam my_camera_name]
location: printer
enabled: True
service: mjpegstreamer
target_fps: 1
target_fps_idle: 1
stream_url: /webcam?action=stream
snapshot_url:/webcam?action=snapshot

`

it works with obico.io but not with my self host.
it seems everything ok on printer side post picture and responses are ok but server dont see image i cant figure it out
here is the log. 

`
2023-09-19 19:46:21,634     DEBUG  obico.moonraker_conn - Sending to Moonraker: 
{'jsonrpc': '2.0', 'method': 'printer.objects.query', 'id': 54698, 'params': {'objects': {'webhooks': None, 'print_stats': None, 'virtual_sdcard': None, 'display_status': None, 'heaters': None, 'toolhead': None, 'extruder': None, 'gcode_move': None, 'fan': None, 'heater_bed': None}}}
2023-09-19 19:46:21,704     DEBUG  urllib3.connectionpool - http://192.168.1.160:3334 "POST /api/v1/octo/pic/ HTTP/1.1" 200 15
2023-09-19 19:46:21,707     DEBUG  obico.webcam_capture - Jpeg posted to server - viewing_boost: True - <Response [200]>
2023-09-19 19:46:21,717     DEBUG  urllib3.connectionpool - Starting new HTTP connection (1): localhost:80
2023-09-19 19:46:21,722     DEBUG  urllib3.connectionpool - http://localhost:80 "GET /webcam?action=snapshot HTTP/1.1" 301 185
2023-09-19 19:46:21,741     DEBUG  urllib3.connectionpool - http://localhost:80 "GET /webcam/?action=snapshot HTTP/1.1" 200 None
2023-09-19 19:46:21,766     DEBUG  urllib3.connectionpool - Starting new HTTP connection (1): 192.168.1.160:3334
2023-09-19 19:46:21,877     DEBUG  urllib3.connectionpool - http://192.168.1.160:3334 "POST /api/v1/octo/pic/ HTTP/1.1" 200 15
2023-09-19 19:46:21,880     DEBUG  obico.webcam_capture - Jpeg posted to server - viewing_boost: True - <Response [200]>
2023-09-19 19:46:21,889     DEBUG  urllib3.connectionpool - Starting new HTTP connection (1): localhost:80
2023-09-19 19:46:21,894     DEBUG  urllib3.connectionpool - http://localhost:80 "GET /webcam?action=snapshot HTTP/1.1" 301 185
2023-09-19 19:46:21,912     DEBUG  obico.moonraker_conn - Received from Moonraker: {'jsonrpc': '2.0', 'result': {'status': {'virtual_sdcard': {'progress': 0.0, 'file_position': 0, 'is_active': False, 'file_path': None, 'file_size': 0}, 'heaters': {'available_sensors': ['heater_bed', 'extruder'], 'available_monitors': [], 'available_heaters': ['heater_bed', 'extruder']}, 'display_status': {'progress': 0.0, 'message': None}, 'heater_bed': {'temperature': 27.02, 'power': 0.0, 'target': 0.0}, 'print_stats': {'info': {'total_layer': None, 'current_layer': None}, 'print_duration': 0.0, 'total_duration': 0.0, 'filament_used': 0.0, 'filename': '', 'state': 'standby', 'message': ''}, 'toolhead': {'square_corner_velocity': 5.0, 'max_accel': 5000.0, 'homed_axes': 'xyz', 'estimated_print_time': 5915.531996166666, 'max_velocity': 500.0, 'print_time': 7.4742871785714255, 'max_accel_to_decel': 4000.0, 'axis_minimum': [-4.0, 0.0, -5.0, 0.0], 'stalls': 0, 'axis_maximum': [235.0, 235.0, 268.0, 0.0], 'position': [110.0, 110.0, 0.0, 0.0], 'extruder': 'extruder'}, 'fan': {'speed': 0.0, 'rpm': None}, 'gcode_move': {'homing_origin': [0.0, 0.0, 0.0, 0.0], 'speed_factor': 1.0, 'gcode_position': [110.0, 110.0, 0.0, 0.0], 'absolute_extrude': True, 'absolute_coordinates': True, 'position': [110.0, 110.0, 0.0, 0.0], 'speed': 1500.0, 'extrude_factor': 1.0}, 'webhooks': {'state': 'ready', 'state_message': 'Printer is ready'}, 'extruder': {'motion_queue': None, 'pressure_advance': 0.022, 'temperature': 26.76, 'power': 0.0, 'can_extrude': False, 'smooth_time': 0.02, 'target': 0.0}}, 'eventtime': 5966.888743969}, 'id': 54698}
2023-09-19 19:46:21,915     DEBUG  obico.server_conn - Sending to server: 
{'current_print_ts': -1, '
`

but my server not shows image

[image]

burak1000, 2023-09-19:
Thank you @kennethjiang 

I solved my problems i will write here to maybe it helps some one.

I using ELEGOO Neptune4 and it use MKS-PI under with ARMBIAN

*my first mistake is i forget the Django Site  settings. it's important
*my second mistake is mks-pi need to update (you can update everyting from fluidd under de settings menu)
*third is elegoo printer firmware comes with webcamd service not loaded you must enable and register it
*fourth in moonraker.conf you must to add webcam settings.

I cant find anyway to way fix janus on this device but now 0.1fps enough for me.

small tip: elegoo neptune 4 has only one usb port but if you open the main board you will see unsoldered 2 x usb2.0 port.
if you can solder a port here you can connect your camera here.

I soldered and i connect it to usb stick♥
[image]

closed 2023-09-19 by burak1000

kennethjiang, 2023-09-19:
Great! Thank you for sharing this tip @burak1000 !

Any chance you can write up steps you took to make Obico work on the neptune 4. I'd like to publish it to our official doc site so that other neptune 4 users can also benefit from it. Thanks again!

burak1000, 2023-09-20:
dear @kennethjiang  
I'm working hard on it for 6 days.  I actually I'm Embedded System Programmer. but i'm  a bit far to linux/debian systems but i love them.

and of course i can share my guide with pleasure.

neptune 4 uses custom version of MKS-PI also it's CPU and GPU is very powerful even than RPI3

only it needs some custom installations and configrations  for work with obico.
also obico-moonraker needs some code changes now I make small changes to run properly janus&FFmpeg. 

When I reach complete working obico server with neptune i will share with you.

Thank you so much for your messages and helps.

kennethjiang, 2023-09-22:
> dear @kennethjiang I'm working hard on it for 6 days. I actually I'm Embedded System Programmer. but i'm a bit far to linux/debian systems but i love them.
>
> and of course i can share my guide with [...]

Great! Thanks again @burak1000