TheSpaghettiDetective/obico-server #820: Error installing OctoPrint plugin: [Errno 2] No such file or directory:

W1ngR1d3r, 2023-09-29:
I tried several times to update my Obico plugin but with no success so I decided to uninstalled it and re-install the plugin.
Unfortunatelly when I try to install the plugin I get the following error:
WARNING: Error parsing requirements for octoprint-obico: [Errno 2] No such file or directory: '/home/pi/oprint/lib/python3.7/site-packages/OctoPrint_Obico-2.3.1.dist-info/METADATA'
ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: '/home/pi/oprint/lib/python3.7/site-packages/OctoPrint_Obico-2.3.1.dist-info/RECORD'

W1ngR1d3r, 2023-09-29:
Finally could solve the problem. I had to connect to the server and delete a file ffmpeg that should have been a folder and empty the OctoPrint_Obico-2.3.1.dist-info folder.
After that I could finally install Obico.

kennethjiang, 2023-10-03:
Glad you figured this out! Thank you for sharing the fix!

closed 2023-10-03 by kennethjiang