TheSpaghettiDetective/obico-server #822: [BUG] email images not working (self-hosted)

ApexArray, 2023-10-02:
## Describe the bug
Email notification images are not working for me (due to Gmail link rewriting). 

This was originally addressed in #43 by attaching the image rather than embedding (if the link was an internal IP address), but that seems to be broken now.

## Potential Fix
I found the following relevant section of code: https://github.com/TheSpaghettiDetective/obico-server/blob/022d84a77527718b11947057b8604ccef03331c7/backend/notifications/plugins/email/__init__.py#L161

There are two problems that prevent this from working as expected:
  - the `requests` module is not imported in this file
  - invalid reference to `context.img_url` (should just be `img_url`)

I am testing a fix now and will submit a PR if it works.

kennethjiang, 2023-10-03:
Thank you so much for catching this bug! Looking forward to your PR!

ApexArray, 2023-10-03:
@kennethjiang Dang, I encountered another issue.

After a few prints, my emails now have multiple images attached. 
[image]

This appears to be caused by improperly setting a default mutable value on line 144, so the attachments variable keeps growing with each new notification:
https://github.com/TheSpaghettiDetective/obico-server/blob/8be559f33362f540e5b8bdc417c288346d3493d5/backend/notifications/plugins/email/__init__.py#L136-L144

This can be fixed with something like this
```python
    def send_emails(self,
                    user: UserContext,
                    ...,
                    attachments: Optional[List] = None) -> None:
        attachments = attachments or []
```

I can open a new PR. Should I reference this issue or open a new one?

kennethjiang, 2023-10-04:
Good catch! You can just reference this issue. I reopened it.

PR #823 (2023-10-03): Fix email image attachments
Fixes #822
closed 2023-10-04 by kennethjiang