TheSpaghettiDetective/obico-server #837: Web with obico-server didn't capture the stream

YaphetS1, 2023-11-15:
Hi there!
I had write an issue in moonraker-obico, but I think the problem is here. Help me pls 🥲 

To avoid duplication, I will attach a link to the problem [#83](https://github.com/TheSpaghettiDetective/moonraker-obico/issues/83)

YaphetS1, 2023-11-16:
Today I'm tried to change resolution of camera capture and change browser to chrome - it's get me more information for debug. But not clarify the situation... Camera resolution reduce to 1280x720 from native 2592x1944. And again the clipper caught the stream, but the obiko did not

YaphetS1, 2023-11-16:
In obico-server i got this in console
[image]

kennethjiang, 2023-11-17:
in our work flow, github issues are opened only for confirmed bugs or feature requests. Again what you described here looks more like a configuration issue. Please come to our discord channel to get prompt help.

closed 2023-11-17 by kennethjiang