TheSpaghettiDetective/obico-server #846: [Feature] Separate "Pause" Notification from "When other event happens"

puterboy, 2023-12-24:
Currently the "When other event happens" notification lumps together: Start, Pause, and Resume.
Yet they are very different.
I don't generally need a notification for "Start" and "Resume" since typically I know that status since I am the one who intitiated or resumed a print -- as such to me, they are the lowest priority type of notification and ones that I want to mute.

In contrast, "Pause" (along with "Filament Change" and "Failure") are the most **critical** types of notification since they require user intervention to respond  -- to me this is the type of notification that I want to be loud and intrusive so that no matter where I am, I know that something either went wrong or needs my personal intervention.

With my limited talents, I can "hack" the code in the relevant plugins to ignore Start/Resume but IMO, the Notification structure should be redesigned so that if a group of notifications are lumped together with no ability to turn them on/off separately, then at least they should have the same general notion of priority.

JuanR3D, 2024-10-10:
Hi! I'd also like this feature. 
In fact, is there a way to configure the notification sound of these type of events (in the mobile app)? For example, when a pause is commanded due to any failure, notify me and execute a alarm sound.

Thanks in advance!

kennethjiang, 2024-10-11:
The feature is already implemented.

closed 2024-10-11 by kennethjiang