TheSpaghettiDetective/obico-server #884: [BUG] Brief bug description here

deboy69, 2024-02-20:
**Note**
Most of the times, an issue you may potentially run into is not a bug. Instead, it is related to configurations, such as firewall rules, incompatible docker version, reverse proxy setups, etc.

If you are not sure, please don't open a bug report here. Instead, you can get the issue solved much faster by seeking help in our support forum: https://obico.io/discord.

Proceed to opening a bug report only if you are sure it is a defect in the code given a correct configuration.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
Docker-compose pull and Docker-compose up -d

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
 - OS: Ubuntu 22.04.4
 - Version Unkown since i cant run it. 
 - Hardware Dell Ubuntu PC

**Additional context**
Hello I get the following error with a docker compose pull and up command.

  File "/usr/local/lib/python3.7/logging/config.py", line 384, in resolve
    found = self.importer(used)
ModuleNotFoundError: No module named 'colorlog'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/logging/config.py", line 543, in configure
    formatters[name])
  File "/usr/local/lib/python3.7/logging/config.py", line 654, in configure_formatter
    result = self.configure_custom(config)
  File "/usr/local/lib/python3.7/logging/config.py", line 469, in configure_custom
    c = self.resolve(c)
  File "/usr/local/lib/python3.7/logging/config.py", line 397, in resolve
    raise v
  File "/usr/local/lib/python3.7/logging/config.py", line 384, in resolve
    found = self.importer(used)
ValueError: Cannot resolve 'colorlog.ColoredFormatter': No module named 'colorlog'
The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "manage.py", line 15, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 357, in execute
    django.setup()
  File "/usr/local/lib/python3.7/site-packages/django/__init__.py", line 19, in setup
    configure_logging(settings.LOGGING_CONFIG, settings.LOGGING)
  File "/usr/local/lib/python3.7/site-packages/django/utils/log.py", line 76, in configure_logging
    logging_config_func(logging_settings)
  File "/usr/local/lib/python3.7/logging/config.py", line 800, in dictConfig
    dictConfigClass(config).configure()
  File "/usr/local/lib/python3.7/logging/config.py", line 546, in configure
    'formatter %r' % name) from e
ValueError: Unable to configure formatter 'console_debug'

LordDarkneo, 2024-06-24:
Hey there!

I just installed the docker on DSM (Synology) and have the exact same issue with colorlog. This missing dependency is not allowing the server_tasks nor server_web to be started in Docker (probably the reason why you dont have the folder yet).

There are plenty of other topics on this "optional" package but I was not able to find a way to install properly this dependancy in the Python virtual of the Obico server (+make it permanent... Cuz I'd like to keep it and not having to reinstall it everytime the doscker is started).

Thanks for your help

kennethjiang, 2024-07-04:
Please use "docker compose up --build -d" to upgrade to see if it'll fix it.

deboy69, 2024-07-05:
Pulled the latest this morning. Seemed to fix the error

closed 2024-07-08 by kennethjiang

LordDarkneo, 2024-07-08:
Sorry for late answer but yes, I confirm it's fixed with the new version (as per our Discord discussion ;) )