TheSpaghettiDetective/obico-server #935: [BUG] UI blank on PC browser

jamincollins, 2024-04-25:
**Describe the bug**
UI blank on PC browser

**To Reproduce**
Steps to reproduce the behavior:
1. clone any repository state including commit 4c4ef680e7b92beba3407a81712fce6b1a73ae88
2. docker-compose up --detach --build
3. access the UI in a PC browser
4. login
5. blank screen

Located the above commit through a `git bisect`.

Reverting the above commit fixes the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Hosting environment (please complete the following information):**
 - OS: Arch Linux
 - Version -- latest
 - Hardware -- virtual machine

**Additional context**
Add any other context about the problem here.

kennethjiang, 2024-04-27:
is the problem solved after you switch to `release` or `pre-release` branch?

jamincollins, 2024-04-30:
yes