TheSpaghettiDetective/obico-server #974: [BUG] Failure Detection is not working

shom, 2024-06-09:
on web ui not Failure Detection.

docker logs obico-server-web-1
**Screenshots**
[image]
**Hosting environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]
 - Hardware [e.g., PC, a Jetson Nano]

**Additional context**
Add any other context about the problem here.

strahd-zarovich, 2024-06-09:
I am having this issue too. Woke up to a lot of Spaghetti today. I do not know how to pull all the logs, but here is what I am working with.

moonraker-obico: v2.0.1-0-g1a527d58
Standalone Obico server one on Linux machine and one on UnRAID server both have the same issue.

I had to hard code my webcam it was not picked up automatically like it was in the past.

No video after the print was manually cancelled, so no time-lapse to look through.

Weirdly my phone app say there is no video when the print was happening, but the server saw the video.

1012302887, 2024-06-11:
I am having this issue too.There is no AI related option on this web page.

kennethjiang, 2024-06-16:
Can you please switch to the `pre-release` branch? https://discord.com/channels/614543405724205137/661368008781987841/1248070919839158343 This issue should be fixed on that branch now.

shom, 2024-06-16:
> Can you please switch to the `pre-release` branch? https://discord.com/channels/614543405724205137/661368008781987841/1248070919839158343 This issue should be fixed on that branch now.

Yes, it worked again after I switched to pre-release yesterday

closed 2024-06-16 by shom