TheSpaghettiDetective/obico-server #975: Module Oauth_provider not found during compose

ChGeorgi, 2024-06-12:
72058e6127e3_obico-server-ml_api-1  | Succeeded!
web-1                               | Traceback (most recent call last):
web-1                               |   File "manage.py", line 15, in <module>
web-1                               |     execute_from_command_line(sys.argv)
web-1                               |   File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
web-1                               |     utility.execute()
web-1                               |   File "/usr/local/lib/python3.7/site-packages/django/core/management/__init__.py", line 357, in execute
web-1                               |     django.setup()
web-1                               |   File "/usr/local/lib/python3.7/site-packages/django/__init__.py", line 24, in setup
web-1                               |     apps.populate(settings.INSTALLED_APPS)
web-1                               |   File "/usr/local/lib/python3.7/site-packages/django/apps/registry.py", line 91, in populate
web-1                               |     app_config = AppConfig.create(entry)
web-1                               |   File "/usr/local/lib/python3.7/site-packages/django/apps/config.py", line 90, in create
web-1                               |     module = import_module(entry)
web-1                               |   File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
web-1                               |     return _bootstrap._gcd_import(name[level:], package, level)
web-1                               |   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
web-1                               |   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
web-1                               |   File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
web-1                               | ModuleNotFoundError: No module named 'oauth2_provider'
web-1 exited with code 1
tasks-1                             | Traceback (most recent call last):
tasks-1                             |   File "/usr/local/bin/celery", line 8, in <module>
tasks-1                             |     sys.exit(main())
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/__main__.py", line 16, in main
tasks-1                             |     _main()
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 322, in main
tasks-1                             |     cmd.execute_from_commandline(argv)
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 499, in execute_from_commandline
tasks-1                             |     super(CeleryCommand, self).execute_from_commandline(argv)))
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/bin/base.py", line 305, in execute_from_commandline
tasks-1                             |     return self.handle_argv(self.prog_name, argv[1:])
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 491, in handle_argv
tasks-1                             |     return self.execute(command, argv)
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/bin/celery.py", line 419, in execute
tasks-1                             |     ).run_from_argv(self.prog_name, argv[1:], command=argv[0])
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/bin/worker.py", line 223, in run_from_argv
tasks-1                             |     return self(*args, **options)
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/bin/base.py", line 253, in __call__
tasks-1                             |     ret = self.run(*args, **kwargs)
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/bin/worker.py", line 258, in run
tasks-1                             |     **kwargs)
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/worker/worker.py", line 99, in __init__
tasks-1                             |     self.app.loader.init_worker()
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/loaders/base.py", line 114, in init_worker
tasks-1                             |     self.import_default_modules()
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/loaders/base.py", line 108, in import_default_modules
tasks-1                             |     raise response
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/utils/dispatch/signal.py", line 288, in send
tasks-1                             |     response = receiver(signal=self, sender=sender, **named)
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/fixups/django.py", line 84, in on_import_modules
tasks-1                             |     self.worker_fixup.validate_models()
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/fixups/django.py", line 122, in validate_models
tasks-1                             |     self.django_setup()
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/celery/fixups/django.py", line 118, in django_setup
tasks-1                             |     django.setup()
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/django/__init__.py", line 24, in setup
tasks-1                             |     apps.populate(settings.INSTALLED_APPS)
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/django/apps/registry.py", line 91, in populate
tasks-1                             |     app_config = AppConfig.create(entry)
tasks-1                             |   File "/usr/local/lib/python3.7/site-packages/django/apps/config.py", line 90, in create
tasks-1                             |     module = import_module(entry)
tasks-1                             |   File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
tasks-1                             |     return _bootstrap._gcd_import(name[level:], package, level)
tasks-1                             |   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
tasks-1                             |   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
tasks-1                             |   File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
tasks-1                             | ModuleNotFoundError: No module named 'oauth2_provider'

mattmcquinn, 2024-06-14:
I have been getting the same error for a while now. Let me know if I can help test in any way.

ChGeorgi, 2024-06-14:
It seems that nobody is interested in a solution for  the problem, neither in discord nor here. I could also write what I tried to fix. My problem occurred after update/ upgrade of os and install of apache2

JaymZZZZ, 2024-06-15:
I fixed it by running the following:

`docker-compose down`
`docker-compose` up --force-recreate --build`

mattmcquinn, 2024-06-15:
Thanks @JaymZZZZ that worked for me! You rock!

kennethjiang, 2024-06-16:
@JaymZZZZ thank you for the tip. I didn't even know the `--force-recreate` option.

Closing this issue now.

closed 2024-06-16 by kennethjiang