TheSpaghettiDetective/obico-server #977: [Feature] Display pictures in print completion emails and Discord notifications

JonahMMay, 2024-06-16:
When pause/resume notifications are enabled, they send an image from the camera as part of the email and Discord message. However, this does not occur for notifications sent when a print has finished.

Can the a photo from the camera be added to this notification?

kennethjiang, 2024-06-19:
Can you attach the email you received when the print was done?

JonahMMay, 2024-07-06:
Here is an email.

[image]

kennethjiang, 2024-07-08:
I can't reproduce this issue. Can you link your printer to the Obico cloud (https://app.obico.io) to see if you have the same issue? If not, it probably has to do with the settings in your self-hosted server.

JonahMMay, 2024-07-09:
I had my printer added originally and it worked. Is there a specific setting or config value I should look for that could be disabling it?

kennethjiang, 2024-07-14:
> I had my printer added originally and it worked. Is there a specific setting or config value I should look for that could be disabling it?

We are always making changes to the server, the app, and the plugin. So not sure how I can address your question easily.