TheSpaghettiDetective/obico-server #980: [BUG] `pre-release` build fails with Traceback

jamincollins, 2024-06-18:
**Describe the bug**

Updated to `pre-release` branch per Discord announcement instructions:
https://discord.com/channels/614543405724205137/661368008781987841/1248070919839158343

Both mobile and web UI fail with a **Server Error (500)**.

Web UI after self hosted upgrade to `pre-release`:
[image]

Mobile UI after self hosted upgrade to `pre-release`:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. cd obico-server
2. git checkout pre-release
3. git pull
4. docker-compose up --build -d

```
$ git remote -v
origin	https://github.com/TheSpaghettiDetective/obico-server.git (fetch)
origin	https://github.com/TheSpaghettiDetective/obico-server.git (push)
```

```
$ git reflog  -1
96129724 (HEAD -> pre-release, origin/pre-release) HEAD@{0}: checkout: moving from release to pre-release
```

Output from web-1 container:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/asgiref/sync.py", line 534, in thread_handler
    raise exc_info[1]
  File "/usr/local/lib/python3.10/site-packages/django/core/handlers/exception.py", line 43, in inner
    response = await get_response(request)
  File "/usr/local/lib/python3.10/site-packages/django/core/handlers/base.py", line 253, in _get_response_async
    response = await wrapped_callback(
  File "/usr/local/lib/python3.10/site-packages/asgiref/sync.py", line 479, in __call__
    ret: _R = await loop.run_in_executor(
  File "/usr/local/lib/python3.10/site-packages/asgiref/current_thread_executor.py", line 40, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.10/site-packages/asgiref/sync.py", line 538, in thread_handler
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/generic/base.py", line 84, in view
    return self.dispatch(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 46, in _wrapper
    return bound_method(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/decorators/debug.py", line 92, in sensitive_post_parameters_wrapper
    return view(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 46, in _wrapper
    return bound_method(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/decorators/cache.py", line 62, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/allauth/account/views.py", line 84, in dispatch
    return super().dispatch(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/allauth/account/mixins.py", line 40, in dispatch
    response = super().dispatch(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/generic/base.py", line 119, in dispatch
    return handler(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/allauth/account/mixins.py", line 59, in get
    response = super().get(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/generic/edit.py", line 144, in get
    return self.render_to_response(self.get_context_data())
  File "/usr/local/lib/python3.10/site-packages/allauth/account/views.py", line 106, in get_context_data
    site = get_current_site(self.request)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/shortcuts.py", line 14, in get_current_site
    return Site.objects.get_current(request)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/models.py", line 61, in get_current
    return self._get_site_by_request(request)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/models.py", line 45, in _get_site_by_request
    SITE_CACHE[domain] = self.get(domain__iexact=domain)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 496, in get
    raise self.model.DoesNotExist(
django.contrib.sites.models.Site.DoesNotExist: Site matching query does not exist.
django.request WARNING  Not Found: /favicon.ico
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hosting environment (please complete the following information):**
 - OS:  Arch Linux
 - Version - fully up to date
 - Hardware - virtual machine, 2 vcpu, 4G RAM

**Additional context**
Add any other context about the problem here.

jamincollins, 2024-06-18:
It appears the root cause for this was not having the FQDN I was using to access the server in the list of sites.

This is a behavior change from the previous (current `release` as of this writing) build.

closed 2024-06-18 by jamincollins

levydanqc, 2024-06-28:
Hi! Still happening from branch `release`.. what would be the fix for this please ?

jamincollins, 2024-06-29:
From the Discord:

> A new release for obico-server. This new release is required for Obico app version greater than 2.5 to stream correctly. To update:

```
cd obico-server
git checkout release
git pull
docker compose up --build -d
```

> This release also has a new site configuration tool that makes the server more secure. If you have run into 500 errors after the upgrade,  please run a site management command to fix it

```
docker compose run web ./manage.py site --fix
```

---

It's the last bit I think you might need.