TheSpaghettiDetective/obico-server #983: Camera no longer working in app

billkenney, 2024-06-20:
I'm not sure why, but my camera feed has stopped working in the app. I'm running obico server self hosted on docker and obico on klipper. The webcam works on the browser, just not on the app. The app also no longer has timelapse videos. But Timelapse and the webcam are working in fluidd. I don't know if there is something in the database that's corrupt or some way to troubleshoot this, but there is nothing in the logs. [image][image]

mon5termatt, 2024-06-23:
Dame here

kennethjiang, 2024-06-23:
Please set the logging level to DEBUG and send the log file to k@obico.io: https://obico.io/docs/user-guides/moonraker-obico/logging-file/

billkenney, 2024-06-24:
> Please set the logging level to DEBUG and send the log file to k@obico.io: https://obico.io/docs/user-guides/moonraker-obico/logging-file/

I sent the logs from a short print and an hour or so after when the printer was idle. I'm self hosting Obico, and I changed my domain to domain.com.

billkenney, 2024-06-25:
I should also note that while I have a Qidi Max 3 printer, I'm running Debian Bookworm and python 3.11, as well as the latest klipper, moonraker, timelapse, etc.

kennethjiang, 2024-06-26:
Got your log file. And sorry for the delay. 

I believe this issue is fixed in the latest release: https://discord.com/channels/614543405724205137/661368008781987841/1255143964248510618

kennethjiang, 2024-06-26:
Please close this issue once you have updated the server and confirmed it's fixed.

billkenney, 2024-06-28:
> Please close this issue once you have updated the server and confirmed it's fixed.

I'm out of town, but will update next week and close the issue if it's resolved. Thanks!

billkenney, 2024-07-04:
> Got your log file. And sorry for the delay. 
> 
> 
> 
> I believe this issue is fixed in the latest release: https://discord.com/channels/614543405724205137/661368008781987841/1255143964248510618

Still not working with the latest release. It works on the web page and on fluidd, but not on the app. 

Here are the logs from the printer:

2024-07-03 20:58:20,570     DEBUG
obico.server_conn - POST
https://obico.domain.com/api/v1/octo/pic/ - 200
2024-07-03 20:58:20,571     DEBUG
obico.server_conn - Jpeg posted to server -
camera name:  - viewing_boost: False - <Response
[200]>
2024-07-03 20:58:31,047     DEBUG
obico.server_conn - POST
https://obico.domain.com/api/v1/octo/pic/ - 200
2024-07-03 20:58:31,048     DEBUG
obico.server_conn - Jpeg posted to server -
camera name:  - viewing_boost: False - <Response
[200]>
2024-07-03 20:58:41,625     DEBUG
obico.server_conn - POST
https://obico.domain.com/api/v1/octo/pic/ - 200
2024-07-03 20:58:41,626     DEBUG
obico.server_conn - Jpeg posted to server -
camera name:  - viewing_boost: False - <Response
[200]>

Here are the logs from the server (there were a ton of the 
/api/v1/octo/printer_events/ 200 15 logs, so i deleted all but one):

172.18.0.1:58018 - - [04/Jul/2024:01:27:32] "WSCONNECTING /ws/janus/2/" - -
172.18.0.1:58032 - - [04/Jul/2024:01:27:32] "WSCONNECTING /ws/janus/2/" - -
172.18.0.1:58042 - - [04/Jul/2024:01:27:32] "WSCONNECTING /ws/web/1/" - -
172.18.0.1:58050 - - [04/Jul/2024:01:27:32] "WSCONNECTING /ws/janus/1/" - -
172.18.0.1:58054 - - [04/Jul/2024:01:27:32] "WSCONNECTING /ws/janus/1/" - -
172.18.0.1:58018 - - [04/Jul/2024:01:27:32] "WSCONNECT /ws/janus/2/" - -
172.18.0.1:58032 - - [04/Jul/2024:01:27:32] "WSCONNECT /ws/janus/2/" - -
172.18.0.1:58042 - - [04/Jul/2024:01:27:32] "WSCONNECT /ws/web/1/" - -
172.18.0.1:58050 - - [04/Jul/2024:01:27:32] "WSCONNECT /ws/janus/1/" - -
172.18.0.1:58054 - - [04/Jul/2024:01:27:32] "WSCONNECT /ws/janus/1/" - -
172.18.0.1:58054 - - [04/Jul/2024:01:27:33] "WSDISCONNECT /ws/janus/1/" - -
172.18.0.1:58000 - - [04/Jul/2024:01:27:33] "POST /api/v1/octo/pic/" 200 15
172.18.0.1:58010 - - [04/Jul/2024:01:27:36] "WSDISCONNECT /ws/web/2/" - -
172.18.0.1:58042 - - [04/Jul/2024:01:27:36] "WSDISCONNECT /ws/web/1/" - -
172.18.0.1:58000 - - [04/Jul/2024:01:27:36] "POST /api/v1/octo/printer_events/" 200 15
172.18.0.1:58050 - - [04/Jul/2024:01:27:36] "WSDISCONNECT /ws/janus/1/" - -
172.18.0.1:58018 - - [04/Jul/2024:01:27:36] "WSDISCONNECT /ws/janus/2/" - -
172.18.0.1:58032 - - [04/Jul/2024:01:27:36] "WSDISCONNECT /ws/janus/2/" - -
172.18.0.1:56580 - - [04/Jul/2024:01:32:51] "POST /api/v1/octo/g_code_files/" 201 286
django.request WARNING  Not Found: /ent/api/personalized_mobile_ui/
api.consumers WARNING  WebConsumer: Closed websocket with code: 1000

kennethjiang, 2024-07-04:
Please set the logging level to DEBUG and send the log file to [k@obico.io](mailto:k@obico.io): https://obico.io/docs/user-guides/moonraker-obico/logging-file/

billkenney, 2024-07-08:
I ended up deleting the program and database on my printer and server, and reinstalling from scratch, and it seems to work fine again. Bummer I lost the print history and Timelapse's, but at least it's working by again

closed 2024-07-08 by billkenney

billkenney, 2024-07-10:
So... the camera on the app works again, but the timelapses do not work. They work on my printer, just not on the Obico server