TheSpaghettiDetective/obico-server #987: [BUG] Portainer can't find manage.py

spusuf, 2024-06-29:
When the image or git is pulled then started using ```docker compose up -d``` the server starts as expected, but using Portainer to launch results in obico-web logs showing:
```
python: can't open file '/app/manage.py': [Errno 2] No such file or directory
python: can't open file '/app/manage.py': [Errno 2] No such file or directory
python: can't open file '/app/manage.py': [Errno 2] No such file or directory
```

And obico-tasks logs show:
```
Usage: celery [OPTIONS] COMMAND [ARGS]...
Try 'celery --help' for help
Error: Invalid value for '-A' / '--app':
Unable to load celery application.
```

However changing the backend volume in ```docker-compose.yml``` from ```- ./backend:/app``` to:
``` - ./backend:/backend ```  works

Adding the line  ```-./backend:/app```does not work.

Is there a reason why the frontend volume points to the frontend directory, but backend points to a directory that doesn't exist in the root folder? Again this is only an issue in Portainer but the fix works for docker compose.

Just want to get a little bit of insight as to why it's set up this way  and what I need to fix in a PR to get working on both platforms.

kennethjiang, 2024-07-04:
"./backend" should be mapped to "/app" in the container. It was for backward compatibility.

goandfix, 2024-08-21:
I have the same problem. Like a work arodun I had to download the github manually and run with portainer.

Wetzel402, 2024-09-18:
I tried this but then see these warning...
```
?: (staticfiles.W004) The directory '/app/../frontend/builds' in the STATICFILES_DIRS setting does not exist.
?: (staticfiles.W004) The directory '/app/../frontend/static' in the STATICFILES_DIRS setting does not exist.
```
and this error...
```
ValueError: Missing staticfiles manifest entry for 'vendor/css/sweetalert2.min.css'
```

geodic, 2024-10-29:
@spusuf, try using the solution I detailed in my comment in [another issue](https://github.com/TheSpaghettiDetective/obico-server/issues/720#issuecomment-2444837807)

Pytonballoon810, 2025-01-29:
What is the solution for the relative path problem? I would like to set up obico as docker as well