TheSpaghettiDetective/obico-server #988: [BUG] Error 500 after installing Obico with docker

microSoftware, 2024-07-01:
Installed Obico with docker as mentioned in your documentation. 
All the instances are running fine as you can see on this screenshot of portainer.
[image]

I adding a mapping port for each of the instance because by default, there were no port mapping.
On the instance "obico-server-web-1", i changed the value for "ML_API_HOST", "REDIS_URL", "INTERNAL_MEDIA_HOST" with my ip. For example "http://192.168.0.104:3333"

and i get an error 500 when going to http://192.168.0.104:3334. I get redirected to http://192.168.0.104:3334/accounts/login/
which show "Server Error (500)"

Zeunas, 2024-07-01:
Same here, I have started from scratch and still got the same error.

MadMurdock123, 2024-07-02:
I have the same issue, without change anything. Pure installation.

kohlerj, 2024-07-03:
Same here!

Look at this one #980 !

S1L3NTANG3L, 2024-07-03:
Same error, log below:
`daphne.cli   INFO     Starting server at tcp:port=3334:interface=0.0.0.0
daphne.server INFO     HTTP/2 support enabled
daphne.server INFO     Configuring endpoint tcp:port=3334:interface=0.0.0.0
daphne.server INFO     Listening on TCP address 0.0.0.0:3334
django.request ERROR    Internal Server Error: /admin/login/
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/models.py", line 39, in _get_site_by_request
    SITE_CACHE[host] = self.get(domain__iexact=host)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 496, in get
    raise self.model.DoesNotExist(
django.contrib.sites.models.Site.DoesNotExist: Site matching query does not exist.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/asgiref/sync.py", line 534, in thread_handler
    raise exc_info[1]
  File "/usr/local/lib/python3.10/site-packages/django/core/handlers/exception.py", line 43, in inner
    response = await get_response(request)
  File "/usr/local/lib/python3.10/site-packages/django/core/handlers/base.py", line 253, in _get_response_async
    response = await wrapped_callback(
  File "/usr/local/lib/python3.10/site-packages/asgiref/sync.py", line 479, in __call__
    ret: _R = await loop.run_in_executor(
  File "/usr/local/lib/python3.10/site-packages/asgiref/current_thread_executor.py", line 40, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.10/site-packages/asgiref/sync.py", line 538, in thread_handler
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 46, in _wrapper
    return bound_method(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/decorators/cache.py", line 62, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/admin/sites.py", line 440, in login
    return LoginView.as_view(**defaults)(request)
  File "/usr/local/lib/python3.10/site-packages/django/views/generic/base.py", line 84, in view
    return self.dispatch(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 46, in _wrapper
    return bound_method(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/decorators/debug.py", line 92, in sensitive_post_parameters_wrapper
    return view(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 46, in _wrapper
    return bound_method(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 134, in _wrapped_view
    response = view_func(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/utils/decorators.py", line 46, in _wrapper
    return bound_method(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/decorators/cache.py", line 62, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/auth/views.py", line 67, in dispatch
    return super().dispatch(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/generic/base.py", line 119, in dispatch
    return handler(request, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/views/generic/edit.py", line 144, in get
    return self.render_to_response(self.get_context_data())
  File "/usr/local/lib/python3.10/site-packages/django/contrib/auth/views.py", line 103, in get_context_data
    current_site = get_current_site(self.request)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/shortcuts.py", line 14, in get_current_site
    return Site.objects.get_current(request)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/models.py", line 61, in get_current
    return self._get_site_by_request(request)
  File "/usr/local/lib/python3.10/site-packages/django/contrib/sites/models.py", line 45, in _get_site_by_request
    SITE_CACHE[domain] = self.get(domain__iexact=domain)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/django/db/models/query.py", line 496, in get
    raise self.model.DoesNotExist(
django.contrib.sites.models.Site.DoesNotExist: Site matching query does not exist.
django.request WARNING  Not Found: /favicon.ico`

S1L3NTANG3L, 2024-07-03:
> Same here!
> 
> Look at this one #980 !

Tried that fix, no luck...
`docker compose run web ./manage.py site --fix
[+] Creating 1/0
 ✔ Container obico-server-ml_api-1  Running                                                                                                                                                                                             0.0s
System check identified some issues:

WARNINGS:
app.User: (auth.W004) 'User.email' is named as the 'USERNAME_FIELD', but it is not unique.
        HINT: Ensure that your authentication backend(s) can handle non-unique usernames.
Sites fixed successfully.
root@Ares:/mnt/user/appdata/obico-server#
`

kohlerj, 2024-07-03:
> Tried that fix, no luck... `docker compose run web ./manage.py site --fix [+] Creating 1/0 ✔ Container obico-server-ml_api-1 Running 0.0s System check identified some issues:
>
> WARNINGS: app.User: [...]

Yep! In the end it did not help as well :(

MadMurdock123, 2024-07-03:
I maybe fix ist:
docker compose run web ./manage.py site --add x.x.x.x
x.x.x.x ist the IP adress from your system, e.g. 192.168.10.50
i add it without any Port.

kennethjiang, 2024-07-03:
For all who are impacted by a similar problem, please update your server, and follow this guide: https://www.obico.io/docs/server-guides/configure/#django-site.

kohlerj, 2024-07-03:
Perfect! Thank you @kennethjiang

S1L3NTANG3L, 2024-07-04:
That fixed it, in the wiki maybe just point the Error 500 to this. It currently points to the smtp setup

kennethjiang, 2024-07-05:
> That fixed it, in the wiki maybe just point the Error 500 to this. It currently points to the smtp setup

 Can you elaborate what you mean?

genehand, 2024-07-06:
If you search the docs for "500", 2 pages come up both related to sending email:

- https://www.obico.io/docs/server-guides/faqs/#i-got-an-http-500-error-when-i-do-sent-test-email-why
- https://www.obico.io/docs/server-guides/advanced/email_guide/#troubleshooting-with-the-test-email

I agree that if this will be a common issue it could help to add it as a tip or to the faq so it's more easily searchable.  Most people (myself included!) probably won't read the full docs including the django config.

S1L3NTANG3L, 2024-07-06:
> If you search the docs for "500", 2 pages come up both related to sending email:
>
>
>
> - https://www.obico.io/docs/server-guides/faqs/#i-got-an-http-500-error-when-i-do-sent-test-email-why
>
> - [...]

Yes this XD
Maybe just put the Django part first that should help a lot 😄

kennethjiang, 2024-07-06:
Good point. will think about how we can surface this info more readily.

closed 2024-07-06 by kennethjiang

miro-ux, 2024-07-16:
[http://localhost:3334/admin/](http://localhost:3334/admin/) works for me but `http://192.168.1.100:3334/admin/` results in same Error 500 and log.